IEAWindSystems/IEA-15-240-RWT #10: License

NilsJohannsen, 2020-02-16:
Hi there,
after a short look at this really great work which is made available here I had immediatly the question:
What about the license?

It was mentioned in some news as "open source", but that is a very broad term and questions occur as a commercial usage for demonstration purpose, for testing and simulation purpose, is it a copyleft license, changes must be published again, ... and so on.

Maybe you could use the common GitHub funcitonality do define a license term for you project.
Thanks and best regards, Nils

evan-gaertner, 2020-02-25:
Hi Nils,

Thank you for pointing out this oversight.  We are discussing it internally and will correct the issue shortly.

Best,
Evan

gbarter, 2020-09-14:
I added the Apache license to be consistent with the IEA Wind 10-MW and this issue is starting to come up more.

closed 2020-09-22 by gbarter