IEAWindSystems/IEA-15-240-RWT #14: Cad Files

HsynCan, 2020-03-17:
Hello i have some issues with the cad files specifically with the blade files it seems it has some problems or i am mistaken please can u check the photo and tell me it is okay or not 
[image]

evan-gaertner, 2020-03-18:
There is indeed an issue with the mesh of the blade in the Solidworks model.  It was generated with an automated tool, which we've been working to improve.  I will upload corrected files when they are ready, which should be about a week from now.

HsynCan, 2020-03-18:
Thank you for reply i will be waiting for new files have nice day

evan-gaertner, 2020-04-01:
I have pushed updated Solidworks models using our updated meshing tool.  Please let me know if you have further questions or problems.  I'm closing this issues for now, unless I hear otherwise.

closed 2020-04-01 by evan-gaertner