IEAWindSystems/IEA-15-240-RWT #19: create quick start guide

evan-gaertner, 2020-03-30:
Better documentation is needed to reduce the number of basic user inquiries and generally improve usability.  Create a "quick start guide" with better instructions on where/how to install required software, what's in the repo and how to use it, etc.

poncemd, 2021-07-04:
That would be ideal, its a great job but dificult to work with

jennirinker, 2021-07-05:
@poncemd could you list a few specific things that were confusing for you? Or some areas where more info could be needed? Evan is unfortunately working elsewhere now, so I'm not sure what he was intending with this issue.

Echolon, 2021-11-03:
So I installed OpenFast & ROSCO as described (I think :) ) (Linux)

We did the ROSCO full clone and installation. But when running `python setup.py install --compile-rosco` it said no CMakeLists.txt found. and yes there are none - that is confusing as we clone the repo of course...

Next thing to do is to run run_model.py?
Well, this came:
```
Traceback (most recent call last):
  File "/home/USERNAME/Documents/Repositories/IEA-15-240-RWT/WISDEM/run_model.py", line 11, in <module>
    from wisdem.rotorse.rotor import RotorSE, Init_RotorSE_wRefBlade
ImportError: cannot import name 'Init_RotorSE_wRefBlade' from 'wisdem.rotorse.rotor' (/home/USERNAME/anaconda3/envs/openfast_env/lib/python3.9/site-packages/wisdem/rotorse/rotor.py)
```
We took a look at the `rotor.py` and saw that there is no `Init_RotorSE_wRefBlade`

ptrbortolotti, 2021-11-03:
Hi,

that WISDEM script is out of date and we'll remove it. Thanks for pointing that out.

If you want to run the model in WISDEM, please refer to 
https://github.com/WISDEM/WISDEM/tree/develop/examples/02_reference_turbines

If you want to run openfast, I'd recommend compiling openfast manually following the instructions available at https://github.com/OpenFAST/openfast

Once compiled, you can call openfast and any .fst file, such as https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT-Monopile/IEA-15-240-RWT-Monopile.fst

Lastly, please check the ROSCO documentation about its compiling. If you fail, please open an issue there. 

Let us know how it goes.

Kind regards,
Pietro

Echolon, 2021-11-03:
Thanks for super fast reaction Pietro!

closed 2022-02-25 by gbarter