IEAWindSystems/IEA-15-240-RWT #20: Wrapper around HAWC2 controller.

SamAct, 2020-05-04:
Do we have any sort of documentation on how to make wrapper around HAWC2 controller?
How did you proceed while creating the dlls for the NREL controller? The hawc2 htc file shows you are supplying a number of inputs to the NREL controller how did you determine the connection between them?

I need to understand this as i ma trying to integrate the NREL discon controllers in HAWC2. Any help or guidance is welcome.

evan-gaertner, 2020-05-06:
Hello,

I assume you're asking about the OpenFAST model I recently uploaded using the DTU Basic Controller.  This was enabled by DTU's work on developing a Bladed Style dll wrapper for the DTU Basic Controller.  This can be found here:
https://gitlab.windenergy.dtu.dk/OpenLAC/BasicDTUController/-/tree/cmake_migration_famen/
See the /src/dtu_we_controller_bladed folder.
These are ongoing projects, with work on publications still in progress.  Unfortunately it is not well documented at this time.

Please let me know if you have further specific questions,
Evan

evan-gaertner, 2020-05-19:
Closing pending further clarification of issue.

closed 2020-05-19 by evan-gaertner