IEAWindSystems/IEA-15-240-RWT #24: 64-bit DLL controller

shermanizer, 2020-05-27:
Right now the controller DLL is in 32 bit and does not seem to work with 64-bit OpenFAST on a 64-bit Windows machine. Would it be possible to also provide a 64-bit version of the DLL controller? Thank you.

gbarter, 2020-09-14:
This one was compiled using Anaconda tools on Windows.  It is 64-bit, but might need Anaconda installed (not sure).  Give it a try and let me know if it works.
https://www.dropbox.com/s/d46fe9lpc2xvhkt/libdiscon.dll?dl=0

deepalisingh286, 2021-01-28:
Ah, I have the same issue!
Could you re-share that file, please?

gbarter, 2021-01-28:
I'm pretty sure this is the same file as what was posted back in Sept:
https://www.dropbox.com/s/5rabdqvzekin0bh/libdiscon.dll?dl=0