IEAWindSystems/IEA-15-240-RWT #26: Trouble regarding installing OpenFAST

TiasaGhosh, 2020-06-29:
Hello everybody,
I have been working with FAST v8.16 adn tried installing OpenFAST for the updated version and 15MW RWT model. However, I always run into issues and I am never able to install the software completely. As I have tried multiple times through various processes (for Windows) described in the guide, there's a basic question from my side regarding installation.
Do we absolutely need to buy Microsoft Visual Studio for OpenFAST? Also, is Intel Parallel Studio (Fortran compiler) subscription required for OpenFAST?
I have had used FAST v8.16 without any further subscription and accounts, so this is bit confusing for me.
thank you for your time.
Best regards,
Tiasa

evan-gaertner, 2020-06-29:
Hello Tiasa,

You might find the prebuilt OpenFAST libraries to be helpful: https://openfast.readthedocs.io/en/master/source/install/index.html#download-binaries

You will still need to compile the ROSCO controller.  There are internal discussions on including compiled binaries for ROSCO as well, but there's no timeline on when this feature will be available.

You can download and use the Community Edition of Visual Studio for free.  Intel Parallel Studio is not required.  You should not need to buy any software license to compile it.  Personally, when I install OpenFAST on Windows, I use cygwin (also free), allowing access to the cmake and make interface for compiling OpenFAST, but this solution might not be best for those new to compiling source code.

Best Regards,
Evan

TiasaGhosh, 2020-07-20:
Hello Evan,
thank you for your response.
I going by the  Cygwin method now. I have installed cygwin and cloned the repositories. 
I am trying to run the regression tests first. I have cloned the repository reg_tests.
However, when I enter the command, 'python manualRegressionTest.py -h', error pops up as:
'  File "manualRegressionTest.py", line 74
    print(strFormat(prefix).format(prefix), strFormat(longestName+" ").format(case), end="", flush=True)
                                                                                        ^
SyntaxError: invalid syntax
'
How should I solve this?
Best regards,
Tiasa

TiasaGhosh, 2020-08-25:
Thank you Evan! I have installed Cygwin and OpenFAST now :)

closed 2020-08-25 by TiasaGhosh