IEAWindSystems/IEA-15-240-RWT #29: Error for 15MW Monopile- BEMT_UnCoupledSolve:

TiasaGhosh, 2020-08-25:
Hello everybody,
I have installed Cygwin and OpenFAST and using them to run 15 MW WT with Monopile. However, I run into an error with the following message:

[image]

Anybody had a similar issue?
Thank you!
Tiasa

ptrbortolotti, 2020-08-25:
Dear Tiasa,
it is hard to understand from this screenshot what might be going on. OpenFAST does not converge, but the reasons might be multiple. Are you calling BeamDyn or ElastoDyn? If you are calling BD, you might want to reduce the time step and try again. Another likely possibility is that the initial conditions do not match the conditions prescribed by the controller at the wind speed that you are feeding through InflowWind. Please check those. One useful test might be to try a simulation without ServoDyn and with the GenDOF off and try again. I did run the model yesterday myself and everything ran smoothly. 
Let me know.
Best regards,
Pietro

TiasaGhosh, 2020-08-26:
Dear Pietro,
thank you for your reasponse. I am using the 15MW with Monopile configuration as given in GitHub (without any changes). The default fetched model has ElastoDyn.
According to your suggestion I switched GenDOF to False and CompServo to 0.
The probem still persits. Just that before FAST aborted at 6e-02 secs of simulation time and this time at 0.1sec.
* OpenFAST 15MW input file *
[image]

* ElastoDyn i/p file*
[image]

* Error message*
[image]

is there anything else I should be trying?
Thank you again!
Tiasa

ptrbortolotti, 2020-08-28:
Dear Tiasa, I upload here a a set of input files that runs for me
[IEA15MW_DLC_monopile.zip](https://github.com/IEAWindTask37/IEA-15-240-RWT/files/5139893/IEA15MW_DLC_monopile.zip) with the latest **dev branch** of OpenFAST.
Let me know if it works on your side. If it does, please compare the two sets and let me know what the difference was. If it does not, please go ahead and try recompiling OpenFAST.
I hope this helps.
Regards,
Pietro

TiasaGhosh, 2020-08-31:
Dear Pietro,
As you suggested, I compiled and installed the latest dev branch of OpenFAST. So the OpenFAST executable is now installed after compiling OpenFAST. I used your set of files for the same.
Now it runs successfully.
The differences I came up with are:
1. the time step as mentioned by you in the previous post
2. some paramters in the linearisation section which should not matter since linearisation is turned off.
3. the Cp-Ct-cp lookup table in .txt file had one difference in line 9. 'Wind speed vector - z axis (m/s)'
It is 10.59 in the older version and 1.0 in the files you provided.

I will now try running the Semi-Sub 15MW case too with this exe. And I will keep posted about this. 
Thank you so much for your help.
regards,
Tiasa

ptrbortolotti, 2020-08-31:
Dear Tiasa,
of the 3 points you list, only #1 can be the relevant one. Good luck with your studies and let us know if you get stuck again. I close this issue now.
Regards,
Pietro

closed 2020-08-31 by ptrbortolotti

TiasaGhosh, 2020-09-04:
Dear Pietro,
I went back to the files I downloaded and compiled before you suggested the Dev branch. And noticed a difference in the ElastoDyn input file for Monopile. The initial rotor speed was 7.55 rpm in the initial ED file and 5.11 rpm in the latest ED file.

Best regards,
Tiasa