IEAWindSystems/IEA-15-240-RWT #31: Wind Turbine Documentation Broken Link

fdchicom, 2020-10-31:
Hello there,

I tried to get access to the wind turbine documentation in the suggested link: https://www.nrel.gov/docs/fy20osti/75698.pdf . However, it appears the link is broken as it fails to download it.

mayankchetan, 2020-11-01:
@fdchicom 
I've had the same issue in the past with documents from the domain (https://www.nrel.gov/docs/*****). You could try opening the link in incognito mode or by clearing the browser cache. Both methods have worked for me.

gbarter, 2020-11-01:
NREL had major network failures this past week, so it is possible that was the source of the problem, depending on when you tried to access the link.  Both links seem to be working for me now, even on a personal machine off of the NREL network.  Give it another try.

@mayankchetan thank you for that tip.

fdchicom, 2020-11-02:
> @fdchicom
> I've had the same issue in the past with documents from the domain (https://www.nrel.gov/docs/*****). You could try opening the link in incognito mode or by clearing the browser cache. Both methods have worked for me.

closed 2020-11-05 by gbarter