IEAWindSystems/IEA-15-240-RWT #32: Blade Structural Properties Mass and Stiffness at the Root

alexmoss97, 2020-11-03:
Hello, PhD Researcher here. I am using your reference blade as a baseline for optimization and I have been calculating the properties at the root. The values given in your spreadsheet are for 10 stations along the blade but zero stiffness and mass at the blade root. I was wondering if there were updated values for the structural properties at the root for me to compare against.

Thanks

gbarter, 2020-11-03:
There must have been a slight bug in the spreadsheet generation code.  Thank you for catching that.

You can find the structural properties in the BeamDyn input files too:
https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_BeamDyn_blade.dat

alexmoss97, 2020-11-03:
Hi again, the link you've shared is a little bit confusing, can you explain what the grid of numbers represents? I'm not familiar with BeamDyn so I don't understand what the format is.

gbarter, 2020-11-03:
Starting on line 11, the sequence is:
Spanwise station (%)
6x6 stiffness matrix
6x6 mass matrix

Hopefully that helps.

closed 2020-11-04 by alexmoss97