IEAWindSystems/IEA-15-240-RWT #33: FAST_InitializeAll:FAST_Init:FAST_ReadPrimaryFile:Invalid logical input for file

SudharsanKalidoss, 2020-11-04:
Hi everyone,

  I'm trying to run IEA-15-240-RWT-Monopile with OpenFAST ver2.4.0. I encountered with invalid logical input error. I tried with precompiled binaries with windows. I also tried running with dev version of the code. I have attached the snapshot of the error. 

Can someone help me with this problem? 

Thanks 
Sudharsan
[image]

gbarter, 2020-11-05:
Thank you for bringing this to our attention.  The OpenFAST team pushed a release that required the input files to be updated.    This should now run using the latest OpenFAST version.

SudharsanKalidoss, 2020-11-05:
Hi gbarter,

  Thank you for the update. I tried running with latest input files. The program started running and showed me different error. 
[image]

Thanks 
Sudharsan

gbarter, 2020-11-05:
I think you have `WindType` set to 4 in the IEA-15-240-RWT_InflowFile.dat file.  Try setting it to 1.

SudharsanKalidoss, 2020-11-05:
I didn't change anything in the IEA-15-240-RWT_InflowFile.dat file. By default, WindType is set to 1. Still, getting the same error.

[image]

gbarter, 2020-11-05:
I looked more closely at your screen shots and you are running TurbSim not OpenFAST.  Is that what you are trying to do?  The default IEA15MW settings use a simple wind field by default.  Also, it looks like you are running this out of the OpenFAST r-test repo.  Just make sure you have the full IEA-15-240_RWT file set.

SudharsanKalidoss, 2020-11-05:
Sorry, I uploaded the wrong error. This is the new error I got.
[image]

gbarter, 2020-11-05:
The ServoDyn input file points to libdiscon.so by default, but that is for Linux machines.  On Windows, edit the ServoDyn input to point to DISCON.dll

SudharsanKalidoss, 2020-11-05:
Thanks for the help. Issue got resolved and ran successfully.

closed 2020-11-12 by gbarter