IEAWindSystems/IEA-15-240-RWT #35: Changing simulation time step

SudharsanKalidoss, 2020-12-08:
Hi everyone,

   I tried to change the DT value in .fst file from 0.005 s to 0.11s. I encountered with an error stating that ElastoDyn input file is not found or improperly formatted. I tried changing DT_out value to 0.11. It didn't work. I have attached the screenshot of the error.  Also I want to know What is Default value for DT_Out.

Can somebody help with this?

Thank you
[image]

gbarter, 2020-12-09:
Your error is in your ServoDyn input.  The ElastoDyn message is just a warning, but that input file is read okay.  The error message is that the ServoDyn time step must be a integer multiple of your simulation time step.  Even if you get past this error, it is possible that a time step of 0.11s will lead to numerical instabilities.  That is a 20x jump from the original value.

SudharsanKalidoss, 2020-12-15:
Hi gbarter,

   As you said, I went past the error but ended with a numerical stability error. I would like to know on what basis the simulation time step is chosen. Will increasing the no of blade nodes allow me to increase the simulation time step? 

Thanks

gbarter, 2020-12-15:
Numerical stability in OpenFAST is a product of both the discretization and the physics involved.  Just changing the number of blade nodes won't necessarily solve your problem.  If you want a more detailed and thorough answer to your question, I suggest you pose it to the [OpenFAST Forums](https://wind.nrel.gov/forum/wind/).

closed 2020-12-15 by SudharsanKalidoss