IEAWindSystems/IEA-15-240-RWT #36: Blade Geometry Issue

WJPeace1, 2020-12-16:
When importing the model into SolidWorks, I am shown the error for the blades as shown in the image titled "error-1", which is: "IEA-15-240-RWT_TipShape_V1_high_resolution has rebuild errors". I am unsure what is causing this. (I have added the "CAD" prefix to the model to sort into appropriate folders").
[image]

Also, as shown in the image titled "error-2", there is a Boss-extrude associated with the blade. What is the purpose of this particular part? Furthermore, is there a method for sealing the bottom of the blade to transform it from a surface body to a solid body?
[image]

Any help would be much appreciated!

Best Regards
William Peace

gbarter, 2020-12-17:
For the blade, which CAD file are you using?  We recently developed a process to produce better quality shapes using OpenSCAD.  I would suggest for the blades you use this [file](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/CAD/IEA-15-240-RWT_blade.stl). 

You showed a picture of the hub, but I didn't see a question about that part, so let me know if I missed something.  We are still working on a better CAD representation for the whole turbine using the WindIO geometry and OpenSCAD.  This will be outer geometry / mold-line only, no internal details.

WJPeace1, 2020-12-17:
Thanks for the prompt reply. The blade CAD I originally used is the SolidWorks file in the original folder. I have directly replaced the blade with the .stl file and will try that now. 
The hub image shows an extruded circle in the centre of the 3 points where the blades are situated. What is the purpose of this part? 

Best Regards
William

gbarter, 2020-12-17:
Hello William,

The hub model should have a flange for connection to the low speed shaft.  If you are seeing other odd elements in the blade attachment cut-outs, that is likely a mistake in the model.  Unfortunately, the CAD modeling was an underfunded side effort for us at the time. We are also in the process of restoring our Solidworks license, so we cannot open up the files currently.  As I mentioned, we are slowly building up some automated CAD model generation with OpenSCAD, so hopefully this can be fixed in the future.  Alternatively, if you want to submit a corrected part, just let me know.

Cheers,
Garrett

WJPeace1, 2021-02-17:
Please find attached a .step file with the updated blade geometry. This geometry does not contain the internal supports. I am in the process of generating a new geometry with these attached.

Cheers
William

[BLADE_SHELL.zip](https://github.com/IEAWindTask37/IEA-15-240-RWT/files/5997088/BLADE_SHELL.zip)

gbarter, 2021-02-18:
Hi William,

Thank you so much for following up and sending back your work!  For now I have included it in a develop branch [here](https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/openfast_dev/CAD).

Can I ask a little more about the steps you used to generate the file?  Did you use the stl-file I had mentioned?  Something else?

Cheers,
Garrett

closed 2022-02-25 by gbarter