IEAWindSystems/IEA-15-240-RWT #37: Issue with blade flapwise mode shapes

abhineet-gupta, 2020-12-16:
The two flapwise mode shapes for the blade defined in "[IEA-15-240-RWT/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_ElastoDyn_blade.dat](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_ElastoDyn_blade.dat)" are likely incorrect.
Both the mode shapes seem like the first bending modes as shown below (normalized).

[image]

(@mayankchetan Our team should follow this issue.)

gbarter, 2020-12-17:
I think you are correct.  I think this was actually a WISDEM bug that swapped the modes incorrectly that was since corrected, but hasn't propagated itself to these input decks yet.  I have in mind a sweeping bug fix that takes advantage of some of our new WISDEM-OpenFAST coupling, but it is not 100% ready yet.  In the meantime, let me know if these mode shapes make more sense to you:
```
---------------------- BLADE MODE SHAPES ---------------------------------------
-0.21369235786154048   BldFl1Sh(2) - Flap mode 1, coeff of x^2
3.4127418390631763     BldFl1Sh(3) -            , coeff of x^3
-4.2972076293819015    BldFl1Sh(4) -            , coeff of x^4
3.1367686457852746     BldFl1Sh(5) -            , coeff of x^5
-1.0386104976050095    BldFl1Sh(6) -            , coeff of x^6
-1.3531134168024368    BldFl2Sh(2) - Flap mode 2, coeff of x^2
3.288984909501215      BldFl2Sh(3) -            , coeff of x^3
-12.333693460704493    BldFl2Sh(4) -            , coeff of x^4
20.393039980621207     BldFl2Sh(5) -            , coeff of x^5
-8.995218012615492     BldFl2Sh(6) -            , coeff of x^6
0.5818691871171588     BldEdgSh(2) - Edge mode 1, coeff of x^2
3.5593034259564034     BldEdgSh(3) -            , coeff of x^3
-10.027031546319972    BldEdgSh(4) -            , coeff of x^4
11.201606033153372     BldEdgSh(5) -            , coeff of x^5
-4.315747099906962     BldEdgSh(6) -            , coeff of x^6

```

abhineet-gupta, 2020-12-17:
Thank you.
These modes do seem correct and match with the expected shapes for a cantilevered structure.
[image]

I understand that after the sweeping but fixes things might change but, meanwhile, can we consider these new mode shapes as the tentative, most likely shapes for the turbine blades?

gbarter, 2020-12-17:
Yes, I anticipate these mode shapes will be the same in the next tagged version of the 15MW ref turbine.  Most of our remaining to-do items are around the nacelle and monopile.

gbarter, 2021-02-09:
It turns out the mode shapes I gave you had a bug in the polynomial fitting for ElastoDyn.  This is fully corrected in the dev branch now.

closed 2021-02-09 by gbarter