IEAWindSystems/IEA-15-240-RWT #38: Running OpenFAST from directory other than the input file directory

jwpeterson, 2020-12-21:
Hello, 

This may be more of a question for OpenFAST developers, but I thought I would try asking here. I am trying to run the UMaine input file from this repository: 

IEA-15-240-RWT/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi.fst

and what I found was that it works fine if I `cd` (I'm on Linux) to the input file directory before running the executable, but it crashes with the following error if I run from any other directory:
```
 **************************************************************************************************
 OpenFAST

 Copyright (C) 2020 National Renewable Energy Laboratory
 Copyright (C) 2020 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

 OpenFAST-v2.4.0-11-g260d9711-dirty
 Compile Info:
  - Compiler: GCC version 7.4.0
  - Architecture: 64 bit
  - Precision: double
  - Date: Dec 16 2020
  - Time: 14:43:48
 Execution Info:
  - Date: 12/21/2020
  - Time: 08:48:12-0600

 OpenFAST input file heading:
     IEA 15 MW offshore reference model on UMaine VolturnUS-S semi-submersible floating platform

 Running ElastoDyn.
 Nodal outputs section of ElastoDyn input file not found or improperly formatted.
 Running AeroDyn.
 Nodal output section of AeroDyn input file not found or improperly formatted.
 Running BEM.
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 1, Blade = 1
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 2, Blade = 1
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 1, Blade = 2
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 2, Blade = 2
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 1, Blade = 3
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 2, Blade = 3
 Running InflowWind.
 Running ServoDyn.
 Running ServoDyn Interface for Bladed Controllers.
 Using legacy Bladed DLL interface.
 Running HydroDyn.
  WARNING:   The random number generator in use differs from the original code provided by NREL.
  This pRNG uses 33 seeds instead of the 2 in the input file.
  Generating incident wave kinematics and current time history.
  Calculating second order difference frequency wave kinematics.
  Reading in WAMIT output with root name
  "/home/jwpeterson/projects/IEA-15-240-RWT/OpenFAST/IEA-15-240-RWT-UMaineSemi/HydroData/IEA-15-240
  -RWT-UMaineSemi".
  Computing radiation impulse response functions and wave diffraction forces.
  Calculating second order difference-frequency force using the full quadratic transfer function.
 Running MoorDyn.
 Warning: invalid output specifier FX.  Starting character must be C or L.
 Warning: invalid output specifier FY.  Starting character must be C or L.
 Warning: invalid output specifier FZ.  Starting character must be C or L.
   Creating mooring system.   3 fairleads, 3 anchors, 0 connects.
    Finalizing ICs using dynamic relaxation.

   t=60  FairTen 1: 2.76325E+06, 2.75740E+06, 2.76559E+06                                             Fairlead tensions did not converge within TMaxIC=60 seconds.

 FAST_InitializeAll:HydroDyn_Init:Waves_Init:VariousWaves_Init: The random number generator in use
 differs from the original code provided by NREL. This pRNG uses 33 seeds instead of the 2 in the
 HydroDyn input file.

  Time: 0 of 10 seconds.
At line 600 of file /mnt/c/Users/egaertne/temp_15mw/ROSCO_toolbox/ROSCO/src/ReadSetParameters.f90 (unit = 89)
Fortran runtime error: Cannot open file '../IEA-15-240-RWT/Cp_Ct_Cq.IEA15MW.txt': No such file or directory

Error termination. Backtrace:
#0  0x7f98719902da in ???
#1  0x7f9871990ec5 in ???
#2  0x7f987199168d in ???
#3  0x7f9871b03c3d in ???
#4  0x7f9871b03f84 in ???
#5  0x7f9865a3b1e3 in __readsetparameters_MOD_readcpfile
	at /mnt/c/Users/egaertne/temp_15mw/ROSCO_toolbox/ROSCO/src/ReadSetParameters.f90:600
#6  0x7f9865a40e8f in __readsetparameters_MOD_setparameters
	at /mnt/c/Users/egaertne/temp_15mw/ROSCO_toolbox/ROSCO/src/ReadSetParameters.f90:555
#7  0x7f9865a37d41 in DISCON
	at /mnt/c/Users/egaertne/temp_15mw/ROSCO_toolbox/ROSCO/src/DISCON.F90:62
#8  0x7f986f75ff56 in ???
#9  0x7f986f760023 in ???
#10  0x7f986f7602b0 in ???
#11  0x7f986f76d498 in ???
#12  0x7f986f7736c9 in ???
#13  0x7f98723a2fa0 in ???
#14  0x7f98723a31da in ???
#15  0x7f98720ba085 in ???
#16  0x7f98720ba48f in ???
#17  0x56261bd0e7ad in ???
#18  0x56261bc0150e in ???
#19  0x7f98715a5b96 in ???
#20  0x56261bc01549 in ???
#21  0xffffffffffffffff in ???
```

The error message
```
Fortran runtime error: Cannot open file '../IEA-15-240-RWT/Cp_Ct_Cq.IEA15MW.txt': No such file or directory
```
seems to indicate that it's looking for a file relative to the directory where I ran the command from rather than relative to the location of the input file. The file in question does indeed exist at the specified relative path (relative to the input file) but not relative to the current working directory. 

Thanks,
John

gbarter, 2020-12-21:
OpenFAST accepts both relative or absolute paths.  Sounds like you want absolute paths.
https://openfast.readthedocs.io/en/master/source/user/aerodyn/input.html

Cheers,
Garrett

jwpeterson, 2020-12-21:
OK, thanks for that link. I think, in order to avoid changing the input files and to keep them portable, I'll stick with the relative paths, but make sure that I run from the same directory as the main (.fst) input file. I am calling OpenFAST using the Python subprocess module, which has the "cwd" parameter for this purpose. Thanks again.

closed 2020-12-21 by jwpeterson