IEAWindSystems/IEA-15-240-RWT #41: Material Property Queries

WJPeace1, 2021-02-14:
Hello,

In the documentation, the composite material orientations are listed as 0 degrees. To which co-ordinate system/axis is the assigned to?

I also can't seem to find any information regarding material properties for the hub? Is there any information regarding the material used? 

Best regards,
William

gbarter, 2021-02-15:
This is a good question.  In the wind turbine ontology project, [WindIO](https://github.com/IEAWindTask37/windIO), the airfoils are pinned orthogonal to the blade reference axis. The composite orientation angles are then passed without modification to the 2-D composite material elasticity tools.  So, the angles are with respect to the airfoil cross section.

With regards to the hub material properties, you are right that this is not well documented in the report or the initial ontology.  We are putting together an update to the definition that call out hub and other drivetrain materials directly in the [yaml ontology file](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/openfast_dev/WT_Ontology/IEA-15-240-RWT.yaml).  In this file, the hub shell material is called out as cast iron:
https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/openfast_dev/WT_Ontology/IEA-15-240-RWT.yaml#L343
with material properties specified as:
https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/openfast_dev/WT_Ontology/IEA-15-240-RWT.yaml#L701

WJPeace1, 2021-02-15:
>  So, the angles are with respect to the airfoil cross section.

Just to clarify, does this mean the axis is from the leading edge to the trailing edge of the aerofoil? Or is it down the length of the blade?

gbarter, 2021-02-15:
LE to TE as you described.

Also looping in @ptrbortolotti just in case . .

ptrbortolotti, 2021-02-15:
Hello William,
thanks for your questions. You may find Figure 1 in this report https://www.nrel.gov/wind/nwtc/assets/pdfs/precomp.pdf useful. Note that we always use 0 angles, so the laminates have the primary direction 1 (and so E11) perpendicular to the blade cross section. If something is still confusing, would you mind rephrasing what still confuses you?
Regards,
Pietro