IEAWindSystems/IEA-15-240-RWT #42: MoorDyn input file typo

kasia129, 2021-02-18:
Hi,

There is likely a typo in the IEA-15-240-RWT-UMaineSemi_MoorDyn input file, line 18: z-position of node 6 should be -200 to place the anchor on the seabed.

Best wishes,
Kasia

gbarter, 2021-02-18:
Hello,

Thank you for bringing this up.  Are you sure you are current with the repository.  I do indeed see -200 on that line:
https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_MoorDyn.dat#L18

kasia129, 2021-02-18:
> CAUTION: This email originated outside the University. Check before clicking links or attachments.

Hi Garrett,

Yes, you are totally right, I must have used some previous version of the file. I saw another issue raised (issue #28) that showed the same problem so I wrongly assumed the typo was not only in my file. I think we can delete my issue in that case. Thanks for noticing and sorry for the misleading post!

Kind regards,
Kasia

https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_MoorDyn.dat#L18<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FIEAWindTask37%2FIEA-15-240-RWT%2Fblob%2Fmaster%2FOpenFAST%2FIEA-15-240-RWT-UMaineSemi%2FIEA-15-240-RWT-UMaineSemi_MoorDyn.dat%23L18&data=04%7C01%7Ckatarzyna.patryniak%40strath.ac.uk%7C4de2d3b408a346da211c08d8d40ab368%7C631e0763153347eba5cd0457bee5944e%7C0%7C0%7C637492489715501317%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=hU465TPgH1cL94YcRyJWpfZDdax%2BBnkH5Ugqm8ORuQU%3D&reserved=0>

closed 2021-02-18 by gbarter

kasia129, 2021-02-18:
Cheers

Closed #42<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FIEAWindTask37%2FIEA-15-240-RWT%2Fissues%2F42&data=04%7C01%7Ckatarzyna.patryniak%40strath.ac.uk%7Cdb2d9127c4fc49494a1f08d8d40bc732%7C631e0763153347eba5cd0457bee5944e%7C0%7C0%7C637492494340608616%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=cB79Hz0NPZrC2d8PpbcGjM4TQPUMOWp1sKUWs8cAzcc%3D&reserved=0>.