IEAWindSystems/IEA-15-240-RWT #49: No Current Influence

danielfrancisconi, 2021-03-29:
Hello,

When I apply a non-zero current on the IEA-15-240-RWT-UMaineSemi case on OpenFAST HydroDyn module it does not affect the results. Also, if everything is fixed and the only applied external force is the current (no wind, no wave...), the hydrodynamic output forces (e.g. HydroFxi) are zero. There are many modelling differences comparing the HydroDyn input file from the Semisub platforms UMaineSemi (IEA-15) and OC4Semi (NREL5MW). For the OC4 case, I can get the hydrodynamic forces due to current, but for the UMaine it is not possible. Is there something missing in the UMaine HydroDyn input file? How to compute the current influence on the UMaine?

Best regards,
Daniel

gbarter, 2021-03-29:
Hello,

You would have to post your input file for us to see what might be going on.  Also, the [OpenFAST forum ](https://wind.nrel.gov/forum/wind/index.php)might be a better spot to get helpful debugging information.  See for instance [this prior thread](https://wind.nrel.gov/forum/wind/viewtopic.php?f=13&t=2306&p=13108&hilit=Current+no+waves#p13108).

Cheers,
Garrett

danielfrancisconi, 2021-03-29:
Hi Garrett,

Thank you for your answer. I've read the thread you suggested but it couldn't help me because I have already turned on and off the 2nd order floating platform forces with no difference on the result. 
And sorry for my lack of precision before, I am referring to the HydroDyn input file of the IEA-15-240-RWT-UMaineSemi case available in the master branch of this git:
https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_HydroDyn.dat
With the only modification of applying any current, for example a Depth-independent current of 1m/s (file attached):
[IEA-15-240-RWT-UMaineSemi_HydroDyn_current.txt](https://github.com/IEAWindTask37/IEA-15-240-RWT/files/6223431/IEA-15-240-RWT-UMaineSemi_HydroDyn_current.txt)
I've posted this issue here because it is related to how the IEA-15 was hydrodynamically modelled. Let me be more clear:
When we look to the HydroDyn file of OC4 SemiSub (from NREL5MW): 
https://github.com/OpenFAST/r-test/blob/01f51597b8f058d6f08ec67d7b9aeb4a2ffccaad/glue-codes/openfast/5MW_OC4Semi_Linear/NRELOffshrBsline5MW_OC4DeepCwindSemi_HydroDyn.dat
the platform members were created (section "Members"), and its drag coefficients written in the "MEMBER-BASED HYDRODYNAMIC COEFFICIENTS (model 3)" section. 
However, for the IEA-15 Hydrodyn file, this section is empty. The only member created (section "Members") refers to section SIMPLE HYDRODYNAMIC COEFFICIENTS (model 1), where the drag coefficients are zero. That is why, I think, the hydrodynamic forces due to current are zero.
So, should one create the members with non-zero drag coefficients in order to observe a current influence on the platform response? Or all the necessary modelling is already implemented?

Best regards,
Daniel

jjonkman, 2021-03-29:
Dear Daniel,

You are correct.  The current-induced loads are zero because the viscous-drag coefficients are zeroed.  HydroDyn itself includes a hybrid combination of a potential-flow solution and a strip-theory solution.  However, this HydroDyn model of the UMaine Semi for the IEA Wind 15-MW reference turbine appears to only include the potential-flow solution.  (The joints and member do not seem to be set up physically.)  You'll have to update the HydroDyn file to set up the strip-theory solution (in terms of joints, members, diameters, and drag coefficients) in order to simulate current-induced loads on this model.

Best regards,

gbarter, 2021-03-29:
@jjonkman Thank you very much for the complete response.

Daniel- A strip-theory friendly version of the Hydrodyn input file with member layout specification is something we are also working towards.  I cannot promise when it would be available though.

jjonkman, 2021-03-30:
Dear @danielfrancisconi,

I haven't done this myself, but I imagine you could infer the drag coefficient as you say.   In surge, the drag coefficient would be something like `AddBQuad*2/(WtrDens*SurgeArea)`, where `SurgeArea` is the submerged cross-sectional area of the floater projected in the surge direction.

Best regards,

closed 2021-05-11 by gbarter