IEAWindSystems/IEA-15-240-RWT #51: Airfoil Polars and Rotor diameter

DanielRibnitzky, 2021-04-16:
Dear Sir or Madame,
I am working as a scientific researcher at university Oldenburg and we want to use the IEA 15MW turbine as a reference in our project. I would have a question concerning the airfoil polars:

The polars located in the folder "...\OpenFAST\IEA-15-240-RWT\Airfoils" are carried out at a Reynolds number of 750000, but the turbine will operate at Reynolds numbers above 10 million. Is there a specific reason why this low Reynolds number was chosen? In the report on page 10 figure 2.3 there are polars shown with a stated Reynolds number of 1e7. Maybe these polars are also available? 

Another question would concern the actual rotor diameter:
In the report in table 1.1 a rotor diameter of 240m and a hub diameter of 7.94m are stated. In table 2.1 the blade length is set to be 117m. Adding the blade length to the hub radius makes 120.97m. So what is the actual rotor radius? 120m or 120.97m?

I appreciate your help and looking forward to your reply.
Yours sincerely 
Daniel Ribnitzky

gbarter, 2021-04-16:
Hello,

We are aware of both of these issues you mentioned.  

1. For the Reynolds numbers in the OpenFAST files, the magnitude has been corrected in a pending update in [this branch](https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/openfast_dev/OpenFAST/IEA-15-240-RWT/Airfoils).  It's the same polar, there was just a mistake in printing out the associated Reynolds number.  Note that with only one Reynolds number polar per airfoil, there is no interpolation in Re-space, so this wasn't causing any problems previously. We will formalize this release when the OpenFAST team releases their pending changes as well.

2. For the rotor diameter, you are correct that there is a slight inconsistency with the report.  The [blade definition in the yaml-file ](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/openfast_dev/WT_Ontology/IEA-15-240-RWT.yaml) is the real authority.  In that file, the blade reference axis z-coordinate terminates at 117.0m with a prebend of 4m.  Total blade arc length is 117.15m and the hub diameter is specified at 7.94m.  This gives a rotor diameter of 2*117.15 + 7.94 = 242.24m (ignoring blade coning).  I will call this out more explicitly in the FAQ for others.

DanielRibnitzky, 2021-04-19:
Thank you very much for the quick and detailed reply!
To me one question would remain:
How were the polars in the folder "OpenFast\Airfoils" derived? Were they simulated with EllipSys 2D as well? 
As the airfoils-folder contains 50 polars and the yaml-file only 8: Are the polars in the airfoil-folder an interpolation of the 8 "yaml-polars"? But I guess these are two different simulations as the Reynolds number also differs. 
Thanks for your help again.

gbarter, 2021-04-19:
Good question.  Yes, the airfoils and polars were interpolated along the span from the original data in the yaml-file.  The underlying polars in the yaml-file are the original ones generated by EllipSys2D.

wonny0503, 2023-07-11:
Dear @gbarter 

I always appreciate your hard work for IEA 15MW Model.

Currently, I am scheduled to use IEA 15MW blade tip section airfoil for CFD simulation.

If you don't mind, I have a question about Reynolds number.

I checked Reynolds number is 3 millions in OpenFAST and 10 millions in IEA 15MW Technical Report.

Is there any reason why you choose 3 millions?

Best regards,
Sangwon

ptrbortolotti, 2023-07-11:
Hello Sangwon,
the Re number is only a placeholder. Since each airfoil only has one set of airfoil polars, no Re effects are included. This set, each set of polars was computed at different Re numbers, you find here the exact number https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/97cf629a622270ed421de98abf99f4314044fcf2/WT_Ontology/IEA-15-240-RWT.yaml#L565
This is a known weakness of the blade modeling where multiple Re numbers should ideally be made available.
Best regards,
Pietro

wonny0503, 2023-07-12:
Dear @ptrbortolotti 

Thank you for your quick and detailed information.

It really helpful to my research.

Have a nice day!

Best regards,
Sangwon