IEAWindSystems/IEA-15-240-RWT #52: Wave Forces on UMaine

danielfrancisconi, 2021-05-18:
Hello,

When I run the IEA-15-240-RWT-UMaineSemi case available in this git, with OpenFAST 2.4.0, the hydrodynamic forces due to waves seem to be wrong. If everything is fixed (degrees of freedom on ElastoDyn = false) and the only applied external force is a regular wave (WaveMod=1), the hydrodynamic output forces (HydroFxi, HydroFyi, HydroFzi) are constants. Is this related to the issue #49? Does the available HydroDyn should not be used for wave force analysis as it happens for the current forces? 

Best regards,
Daniel

jjonkman, 2021-05-19:
Dear @danielfrancisconi,

Are you using the HydroDyn module of the UMaine semisubmersible available from here: https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_HydroDyn.dat?  I'm not the one who put this model together, so, I'm not sure why it was set up the way it was, but this model is set up without any strip-theory solution and with only the hydrostatic and radiation features of the potential-flow solution enabled.  That is, `PotMod` = 1 (potential-flow enabled) with `ExctnMod` = 0 (wave-excitation _disabled_) and `RdtnMod` = 1 (wave-radiation added mass and damping enabled).  If you want to enable wave-excitation loads from potential flow, you should set `ExctnMod` = 1 (wave-excitation enabled through discrete Fourier transforms).

Best regards,

gbarter, 2021-05-19:
Thank you both for identifying the problem and the solution.  I have activated the wave-excitation in the [branch with the pending update](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/openfast_dev/OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_HydroDyn.dat).

Specifying the geometry to enable strip theory calculations will come in a future update, but it is definitely on the list.

danielfrancisconi, 2021-05-20:
Thank you @jjonkman,

This was exactly what I needed! And thank you @gbarter for the update and putting the strip theory on the list.

Best regards,
Daniel

closed 2022-02-25 by gbarter