IEAWindSystems/IEA-15-240-RWT #54: Discon.dll

hgropelli, 2021-06-04:
What is the difeerence between the Discon.dll provided by OpenFAST V2.5.0 and IEA-15-240-RWT? Which i should use?

nikhar-abbas, 2021-06-04:
Hi @hgropelli,
I presume you are refering to the DISCON.dll file that is compiled in the folders [here](https://github.com/OpenFAST/r-test/tree/adffb7e3baa3ef4b8b2ea4b777ba6b23b8048069/glue-codes/openfast/5MW_Baseline/ServoData) when you install OpenFAST v2.5.0.

This DISCON.dll is specific to the NREL5MW wind turbine, and follows the logic described in Section 7 of the [definition of the NREL 5-MW wind turbine](https://www.nrel.gov/docs/fy09osti/38060.pdf).

The DISCON file distributed with the IEA-15MW wind turbine use the [ROSCO](https://github.com/NREL/ROSCO) controller. This is, in many ways, an updated version of the NREL5MW wind turbine, and includes more modern capabilities. You can read about the logic behind ROSCO in the WES pre-print [here](https://wes.copernicus.org/preprints/wes-2021-19/wes-2021-19.pdf).

hgropelli, 2021-06-04:
Actually i refer to DISCON.dll in openfast_v2.5.0_binaries\DISCON_DLLS\64bit which is available in
(https://github.com/OpenFAST/openfast/releases/download/v2.5.0/openfast_v2.5.0_binaries.zip).

Then the correct way to run IEA 15MW U Maine Semi simulation is to mantain the DISCON.dll included in the first release (https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/ServoData)
and for windows include ​libdiscon.dll from rosco v2.2.0 disponible in (https://github.com/NREL/ROSCO/releases/download/v2.2.0/libdiscon.dll) at ServoData?

Obs: I'm using the Openfast v2.5.0 pre compiled binaries for windows (https://github.com/OpenFAST/openfast/releases/download/v2.5.0/openfast_v2.5.0_binaries.zip

nikhar-abbas, 2021-06-07:
Hi @hgropelli,

The DISCON.dll file that comes with the compiled binaries is the NREL5MW specific controller, as mentioned in my previous response.

You are correct that you should use the windows `libdiscon.dll` available at https://github.com/NREL/ROSCO/releases/download/v2.2.0/libdiscon.dll. 

This is the most recent release of the ROSCO controller. If you wish to stay up-to-date with ROSCO, I recommended "watching" the ROSCO git repository to be notified of updates.

Cheers,
Nikhar

closed 2021-06-15 by hgropelli