IEAWindSystems/IEA-15-240-RWT #58: Rigid Body Free Decays

liongct, 2021-06-28:
Hello,

I'm trying to reproduce the free-decay simulations as shown in the Technical Report, July 2020, to validate if I'm using the OpenFAST model correctly. 

I’ve managed to reproduce all DOF except for pitch. It was stated that the free-decay simulations were performed under still wind environment with aerodynamics drag effect included and the blade was orientated such that the drag is minimal. So, different wind speed and blade pitch angle were tested and the closest result obtained is for 30 m/s steady wind and 90 deg blade pitch. The result is compared to the Technical Report as shown below including a free-decay without wind. It seems that the natural period is matching but not the damping.

I would like to clarify the wind speed and blade pitch angle used to produce the results in the Technical Report. Also, is there any other considerations in addition to wind speed and blade pitch angle?

[image]

Thanks.

Best regards,
CT

gbarter, 2021-06-29:
Hello CT,

Bear with us a bit.  It will be a few weeks until the right folks are back from vacation and can take a look at this.

liongct, 2021-06-30:
Hi @gbarter,

Sure. Thank you for the attention.

Best regards,
CT

andreowhite, 2022-04-04:
Hi @gbarter:
I have carried out a free decay test to determine the natural frequencies of the VolturnUS-S system using the OpenFAST files that I downloaded from GitHub.  I have used the same initial conditions from the paper by Allen et al (2020). However, I only obtain the natural frequencies shown in said paper if I modify the initial conditions in ElastoDyn as follows:
1.5 m PtfmSurge -
1.5 m PtfmSway
1.5 m PtfmHeave
0.25 m PtfmRoll
0.25 m PtfmPitch
1.5 m PtfmYaw
This yielded the following natural frequencies:
Surge - 0.007
Sway - 0.007
Heave - 0.049
Roll - 0.036
Pitch - 0.037
Yaw - 0.011
This the practically the same as shown in the paper by Allen et al (2020) except the pitch natural frequency is 0.036 instead of 0.037.
The initial conditions in the paper by Allen et al (2020) are:
20 m PtfmSurge -
20 m PtfmSway
4 m PtfmHeave
10 m PtfmRoll
10 m PtfmPitch
10 m PtfmYaw

This yielded the following natural frequencies:
Surge - 0.0075
Sway - 0.0096
Heave - 0.049
Roll - 0.032
Pitch - 0.036
Yaw - 0.011
Only the pitch, heave and yaw natural frequencies are similar.
Can you please say why the natural frequencies using initial conditions shown in paper by Allen et al (2020) does not match the ones that I obtained given that I used the said conditions?

Thank you.

Regards,
AOAW

mattEhall, 2022-04-04:
Hi Andre,

I think the natural frequencies that are observed from a free decay test really depend on how you make the measurement, so it is difficult to make a high-precision comparison between different free decay tests unless you use the exact same method for counting what the natural period is from the results. That source of differences is in addition to differences caused by starting at different initial offsets. Determining natural periods of a nonlinear system is never going to be perfect unless you look at exactly the same amplitudes with exactly the same method. 

Your results seem quite close to me, except for sway. Maybe it would be worth checking your model (or natural frequency measurement method) some more in the sway direction to figure out why that difference is so big. But otherwise, things look pretty good. If you really wanted an exact match with what's been published, you would probably need to contact Chris Allen and ask for his exact natural period measurement method so you can replicate it (but this strikes me as more effort than likely useful).

Best,
Matt

andreowhite, 2022-04-04:
Noted @mattEhall . Many thanks for the feedback.

Regards,
AOAW

closed 2024-07-11 by gbarter