IEAWindSystems/IEA-15-240-RWT #64: Definition of DLCs in IEA Wind TCP Task 37 report

Wadbled, 2021-09-27:
Hello, 
I am trying to reproduce the semi submersible defined in the IEA Task 37, Definition of the US Maine VolturnUS-S platform in Orcaflex.  
In the table of DLCs, Table 12, in particular DLCs 6.1 and 6.3 there is a setting of yaw: 
+/-8° and +/-20°
Where does this yaw apply to? I understood it is a turbine misalignment but in the case of +/-20°, Orcaflex cannot run it as the turbine becomes unstable. Am I misunderstanding something? 
Best regards,
Xavier

gbarter, 2021-09-29:
Hello Xavier,

The Orcina team is actually in process of developing an official version of the IEA Wind 15-MW reference wind turbine on the VolturnUS-S floating platform.  You can check in with Alan Ross (whom I think you know already) for status and timeline of that effort.

With regards to your DLC 6.x question, the yaw values you mentioned are the angle between the incoming flow direction and the angle the rotor is pointing to, essentially the turbine misalignment you mentioned.  A Google image search for "wind turbine yaw angle" will give you many diagrams if that helps.  DLC 6.x is for extreme cases occurring when the rotor is parked.

A number of multibody codes have difficulty when it comes to simulating high yaw offsets due to numerical limitations.  I cannot speak to your Orcaflex model, but we see the same thing in our OpenFAST simulations at yaw angles of 20-40 degrees.  More information in the OpenFAST forums on this [here](https://wind.nrel.gov/forum/wind/viewtopic.php?f=3&t=1219&p=13703).

Cheers,
Garrett

Wadbled, 2021-09-29:
Thank you for the quick clarification Garrett, and information on Orcina work. 
Could you clarify as well if, in the yaw misalignment DLCs with parked rotor, the blades are idled relative to the wind direction or the rotor orientation? Would the blades oriented with the edge facing the same azimuth as the rotor or with the edge facing the wind? 
Cheers, 

Xavier

gbarter, 2021-09-29:
Blades are idled in the rotor frame.  With a yaw misalignment it is usually not possible to feather the blades with respect to the wind.

closed 2022-02-25 by gbarter