IEAWindSystems/IEA-15-240-RWT #67: libdiscon.dylib for the IEA15MW UMaine Semi-Sub Model

QusayHawari, 2021-10-12:
Hi,

I am working on MACOS, and wanted to ask if there's a libdiscon.dylib 64-bit for thr semi-sub model available, or do I have to generate it myself via cmake?

Thanks in Advance,
Qusay

gbarter, 2021-10-12:
Hello Qusay,

There is a mac-compiled version of ROSCO associated with our pending updates in the development branch:
https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/openfast_dev/OpenFAST/IEA-15-240-RWT/ServoData

See the Release Notes in that branch for further description of changes there.

Cheers,
Garrett

gbarter, 2021-10-12:
@dzalkind does this look familiar to you?

gbarter, 2021-10-12:
If you consistently use the files in the develop branch, those should be consistent with the latest OpenFAST

QusayHawari, 2021-10-12:
Hi,

Thanks for the changes you did!

I applied said changes to the input file, and I got this error coming from Aerdyn15:

 FAST_InitializeAll:AD_Init:ValidateInputData:When AFAeroMod=2, UAMod must be 4 for linearization.
 Set AFAeroMod=1 or UAMod=4.

 OpenFAST encountered an error during module initialization.
  Simulation error level: FATAL ERROR

They were set to 1, and 3 for AFAeroMod and UAMod respectively. 

I changed AFAeroMod=2 and UAMod=4, but got the same exact error.

Thanks in advance,
Qusay.

dzalkind, 2021-10-13:
If `CalcSteady` is `False`, the linearizations will occur at the specified times.  Otherwise, OpenFAST searches for a steady-state solution before linearizing.

dzalkind, 2021-10-21:
Are you using BeamDyn in the simulation?  It helps to look at the OpenFAST outputs and see if there is convergence to steady-state.  If it's close enough, you can try different `TrimTol` values. 

@ptrbortolotti, have you linearized this model with BeamDyn (and I'm assuming blade DOFs) on?  If so, are 2000 seconds enough for convergence, and what value for `TrimTol` do you typically use?

lqw075, 2023-05-31:
> Ah, I see.
> 
> Working perfectly now.
> 
> Thanks!

Hi @QusayHawari 
I'm learning how to linearize "IEA-15-240-RWT-UMaineSemi". Looks like you've succeeded. Congratulations! I'd like to know the specific operation procedure.
Thanks a lot.
lqw