IEAWindSystems/IEA-15-240-RWT #74: Mass and Inertia Values for the IEA-15-240-RWT-UMaineSemi Model.

QusayHawari, 2021-12-09:
Hi,

I was looking over the documentation of the IEA15MW model, and I could not find some specific parameters defined in the ElastoDyn input file. 

Such as the following:

-HubMass
-GenIner
-NacMass
-NacYIner
-YawBrMass

I noticed a difference between the given values in the ElastoDyn files between the master , and dev branches. Which ones are more accurate? and where can I find the true values?

Thanks,
Qusay.

gbarter, 2021-12-09:
> CAUTION: This email originated from outside of NREL. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi Qusay,

The original nacelle mass accounting had a number of mistakes (see comments here<https://github.com/IEAWindTask37/IEA-15-240-RWT/issues/30> and here<https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/openfast_dev/ReleaseNotes.md>).  The new OpenFAST files in the develop branch are correct and consistent with the mass properties in the tabular file here<https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/openfast_dev/Documentation>.  We are also a day or two away from the develop branch being promoted into the main branch.

Cheers,
Garrett

QusayHawari, 2021-12-09:
Great! Thank you for the clarification.

I will use the values in the dev branch then:

     73376   HubMass     - Hub mass (kg)
    1038341   HubIner     - Hub inertia about rotor axis [3 blades] or teeter axis [2 blades] (kg m^2)
    1836784   GenIner     - Generator inertia about HSS (kg m^2)
     645370   NacMass     - Nacelle mass (kg)
   24163711   NacYIner    - Nacelle inertia about yaw axis (kg m^2)
      28280   YawBrMass   - Yaw bearing mass (kg)

Regards,
Qusay.

liongct, 2021-12-10:
Hi Garrett,

On the same topic, I'm looking at the mass properties in the tabular file and in the ElastoDyn file. The mass for nacelle and hub seems to be different. Could you please clarify?

[image]

Thank you.

Best regards,
CT

gbarter, 2021-12-10:
Thanks for catching this!  We have been tweaking some final items ahead of pushing out the design.  The spreadsheet is auto-populated, but the ElastoDyn update requires a few more steps.  I have aligned the values now.

liongct, 2021-12-13:
Hi Garrett,

Thanks for the clarification!

Best regards,
CT

closed 2022-02-25 by gbarter

gbarter, 2022-10-07:
Thank you.  Your suggestion to add comments to the FAQ on the status of nacelle properties is a good one and overdue.  I added this [entry](https://github.com/IEAWindTask37/IEA-15-240-RWT/wiki/Frequently-Asked-Questions-(FAQ)#is-it-me-or-do-the-nacelle-mass-properties-seem-inconsistent).  

For the nacelle mass, the inconsistency is the difference between what OpenFAST and WISDEM consider as part of the "nacelle" (whether the yaw system is included or not), which you pointed out in the revision.  The old report values in Table 5-1 should not be considered at all.