IEAWindSystems/IEA-15-240-RWT #75: Unusual evolution of idling aero loads with yaw misalignment

R-Quancard, 2021-12-15:
Hi,

I am trying to assess the basic aerodynamic thrust loads obtained with the OpenFAST model of the 15MW RWT, in idling conditions and with the wind arriving from different directions (with a fixed yaw so that I am simulating yaw misalignments from 0° to +/-180°). I subtract the part of the load that's due to the rotor's weight, in order to look at aerodynamic loads only. I am looking at the projection of those loads along the rotor axis.

The results obtained are unusual - I say this after checking on various other WTG models, all of which gave results similar to a sine wave: with the IEA-15-240-RWT, these aerodynamic loads become negative as soon as the yaw misalignment reaches a few degrees (around 5°) and go to a minimum around 30°. They go up and down while remaining negative except in the close vicinity of 0°.

I have translated the model to Bladed and obtained a very similar response. Disabling blade flexibility has a negligible impact, so I can at least say that this is not due to an erroneous setup of blade structural properties.

Could you please help me understand this unusual behaviour?

Thanks,
Romain

jjonkman, 2021-12-15:
Dear @R-Quancard,

Can you share a plot of what you are describing?  I have a hard time visualizing the result you are describing.

And just to confirm, presumably you are modeling the idling rotor with:

- the generator DOF enabled and all blades feathered to 90 degrees in ElastoDyn
- the wake/induction and unsteady airfoil aerodynamics models disabled in AeroDyn, and
- the controller disabled in ServoDyn

Is that correct?

Best regards,

R-Quancard, 2021-12-15:
Dear @jjonkman,

Thanks for such a prompt reply.
I confirm that we followed the recommendations for idling simulations, including the 3 points you mentioned.

You will see below three plots:
- A rough representation of what we obtained in OpenFAST, with RotThrust from which we subtracted the weight projected on the tilted rotor axis
- The results obtained with a Bladed model of the same turbine, looking at rotor loads at the hub without the rotor weight contribution (here the wind is always in the same direction but the yaw angle varies); the black curve is with rigid blades, the red one with flexible blades
- Comparable results obtained on Bladed with a model based on the DTU 10MW; this curve looks similar to several others we generated using other WTG models

[image]

[image]

[image]

Thanks for your help,
Best regards,

Romain

jjonkman, 2021-12-15:
Dear @R-Quancard,

Thanks for clarifying.  A couple more clarifying questions:

- Are you running with steady, uniform wind input and plotting the mean response?  Or are you running with turbulence, in which case, what are you plotting?
- When you say you make the blades rigid, are you referring just to the blades, or you are disabling all structural DOFs in the model?

In the last plot for the DTU 10-MW, I don't really understand what you are showing.  With the blades feathered to 90 degrees, I would expect the rotor thrust to be near zero at 0, 180, and 360 degrees and maximum at 90 and 270 degrees yaw angle.  Please clarify.

In your plots for the IEA 15-MW, I'm not sure I fully understand what I'm seeing yet (the answers to my questions above will hopefully help me).  But with structural flexibility enabled, I would expect to see a blade edgewise / tower side-side instability for certain yaw angles (about 20-40 degrees and 320-340 degrees, which would occur due the angle of attack being in a region of a negative lift slope.  This has been noticed for many other turbines, e.g. as discussed for the NREL 5-MW baseline turbine on our forum here: https://wind.nrel.gov/forum/wind/viewtopic.php?f=3&t=1219&p=13703.  I'm not sure I understand the "wiggles" between 60-300 degrees yaw, but if the wind is steady, perhaps the rotor is settling out at different azimuth angles at different wind speeds.  Normally, these load cases would be run with turbulent wind inflow, where the response would be analyzed statistically (in terms of min, mean, max, etc.).

Best regards,

R-Quancard, 2021-12-16:
Dear @jjonkman,

Like I said we are looking at the projection of the aerodynamic loads along the rotor axis, which in OpenFAST is done by taking the variable RotThrust and subtracting the part of the weight it contains (using the sine of the shaft's tilt angle). If I'm not mistaken, this RotThrust variable is a projection along the shaft axis, regardless of the wind direction. In Bladed, we also look at the force on the hub directed along the shaft, that includes all the loads on the rotor (here we removed the weight components as well).

It seems to me that the rotor thrust you say you were expecting corresponds to the aerodynamic loads projected **along the wind direction** instead of **along the shaft axis**; to me the DTU 10MW plot seems to make sense, with zero loads when the wind direction is perpendicular to the shaft axis along which we are projecting our loads (90 and 270°), a maximum with a head wind (0°) and a minimum with a tail wind (180°).

Regarding the two questions you ask:
- The OpenFAST plot comes from turbulent wind simulations where the variable was averaged out. The Bladed plots come from a specific option that calculates steady loads in parked conditions (here with 90° blade pitch and a fixed 0° azimuth - with one blade up).
- When I say we tested with rigid blades, I do mean that only the blades are made rigid for comparison; but the initial study case (focused on the rotor) was already with a rigid tower (and since this specific test was done with Bladed, it is also a situation with a fixed rotor azimuth position).

It is true that we might not expect a completly smooth response (as you said because of instabilities), but when looking at mean values it seems more natural to me to obtain the DTU 10MW's results than the IEA 15MW's. At least, I am very surprised by the values obtained for 8° yaw mislignment for example: the wind is still very close to being aligned with the rotor axis, and yet the variable we are looking at is negative, which means the RNA is not pushed towards the back of the nacelle but on the contrary, to the front of the rotor. This is the result that tipped our curiosity and led us to outputting these plots - and in the end we are also surprised by the rest of these curves.

Thanks for your help,
Best regards,

Romain

R-Quancard, 2021-12-16:
Dear @jjonkman,

First, my apologies: there was a mistake in the pitch angle used to obtain the DTU 10MW plot in my previous message; you will see that it was replaced by a new plot with two "inner waves" that weren't there before.

Like I said we are looking at the projection of the aerodynamic loads along the rotor axis, which in OpenFAST is done by taking the variable RotThrust and subtracting the part of the weight it contains (using the sine of the shaft's tilt angle). If I'm not mistaken, this RotThrust variable is a projection along the shaft axis, regardless of the wind direction. In Bladed, we also look at the force on the hub directed along the shaft, that includes all the loads on the rotor (here we removed the weight components as well).
It seems to me that the rotor thrust you say you were expecting corresponds to the aerodynamic loads projected along the wind direction, and not along the shaft axis as we are representing it here.

Regarding the two questions you ask:
- The OpenFAST plot comes from turbulent wind simulations where the variable was averaged out. The Bladed plots come from a specific option that calculates steady loads in parked conditions (here with 90° blade pitch and a fixed 0° azimuth - with one blade up).
- When I say we tested with rigid blades, I do mean that only the blades are made rigid for comparison; but the initial study case (focused on the rotor) was already with a rigid tower (and since this specific test was done with Bladed, it is also a situation with a fixed rotor azimuth position).

In the light of the corrected DTU 10MW plot, the IEA 15MW curve does not look that different - although it is still more "wiggly" as you put it yourself. Note that when testing with a WTG model based on the NREL's 5MW reference (see plot below), we obtain quite a different curve shape which is closer to what we saw on some other WTG models (industrial ones).
But anyway, our new corrected tests (including the DTU 10MW one) look more similar to the IEA 15MW's, so the differences don't look like much to worry about and are probably only due to blade designs that are quite different.

Thanks for your help,
Best regards,

Romain

_NREL 5MW:_
[image]

jjonkman, 2021-12-16:
Dear @R-Quancard,

OK, thanks for the clarification and the correction.  I'm glad the results make more sense to you now.

I agree with your explanation of the ElastoDyn RotThrust output.  Please note that if you want the pure aerodynamic thrust (also projected along the shaft axis), without having to subtract out the rotor-weight component, you can use AeroDyn's RtAeroFxh output instead of RotThrust.

I would be curious to see if making the wind turbine completely rigid (disabling all structural DOFs) would result in a change to the behavior; this would certainly eliminate any potential for an aero-elastic instability effecting the results.

Best regards,

closed 2022-02-25 by gbarter