IEAWindSystems/IEA-15-240-RWT #78: Linearizing IEA15 MW at Below Rated.

QusayHawari, 2022-01-24:
Hi,

I am trying to linearise the IEA 15MW UMiane model in the below rated region. 

 I noticed that in the ServoDyn file in OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_ServoDyn.dat 
the following parameters were set as shown below(with VSContrl= 1):

---------------------- SIMPLE VARIABLE-SPEED TORQUE CONTROL --------------------
6.0                                   VS_RtGnSp   
19947760.56251191       VS_RtTq     
437898.87522915896     VS_Rgn2K    
10.0                                 VS_SlPc     

I am not sure why VS_RtGnSp is set at 6rpm and not at 7.55rpm, which is the rated speed for the model. I tried setting it to the rated, but I received the following error:

FAST_InitializeAll:ValidatePrimaryData:VS_Rgn2K*VS_RtGnSp^2 must not be greater than
VS_RtTq.

I am guessing that some of the three other inputs should be changed when setting VS_RtGnSp to the rated?

One other point I am not sure about. 

In the .fst file, I set TrimCase to 2 since I am linearising at the below rated region. 
Going to the ElastoDyn file, I noticed that if I give an initial value for the BlPitch other than 0, OpenFAST linearises, but at that fixed initial value, which is not accurate as it will generate the wrong operating points for the GenPwr, GenTq, and GenSpd. The screenshot below shows that the linearization was performed at 1 deg (which is the same value as BlPitch) at 9m/s below rated speed, where it should have performed it at 0 deg.

[image]

Finally, is there a way where I can find more accurate values for the operating points (in the below rated region) other than those given in the definition of the turbine, as the graph is not very precise?

[image]

Thanks in Advance,
Qusay.

dzalkind, 2022-01-24:
Hi @QusayHawari,

We set `VS_RtGnSp` below the rated generator speed so the torque control is not active when the blade pitch is being trimmed in above-rated wind speeds.  I have settled on 90% of rated as a reasonable value, so I'd set that to 6.8040 rpm.

In [the ROSCO input for this model](https://github.com/NREL/ROSCO/blob/9646dbdab604471477c40c33023ec776a9ac9ec0/Test_Cases/IEA-15-240-RWT-UMaineSemi/ServoData/DISCON-UMaineSemi.IN#L58), we use a slightly different value of the `VS_Rgn2K`, which translates to 3.8314e+05 in the units of the ServoDyn input.

The behavior you witness when using TrimCase = 2 is normal behavior.  If you want to set a constant blade pitch in below-rated TrimCase = 2 linearizations, you have to set the initial blade pitch in ElastoDyn.  You can find rotor-only operating points [here](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/Documentation/IEA-15-240-RWT_FineGrid_tabular.xlsx), but that will not include any tower/blade deflection.  The best way, in my experience, to find the operating points of the linear case you are trying to run is to run the nonlinear model in a closed-loop simulation with the same DOFs you will use in the linearization process. 

I'll leave this open until I update the ServoDyn input.  Can you let me know whether those values I have provided work for you?

Best, Dan

Qusay-Hawari, 2022-01-24:
Hi,

Thanks for you reply, and sending the. excel sheet. 

I saw in the same repository that there were generated .lin files at 5m/s, so I tried to perform my on by setting the following based on your suggestion:

- From the excel sheet, the blade pitch at 5m/s was 2.72 deg, so I set that as the initial condition in the ElastoDyn file.
- I set VS_Rgn2K = 3.8314e+05 *(pi/30)^2 to be in Nm/(rpm)^2 as given in the ServoDyn file. (Note in the DISCON.IN, the unit for VS_Rgn2K is in Nm/(rad/sec)^2).
- I set VS_Rgn2K to 6.8040 rpm.

What I got.
[image]

The .lin file in the repo.
[image]

Excel sheet 

[image]

The results seem to match, but shouldn't the minimum rotor speed be 5 rpm?

Thanks in advance,
Qusay.

dzalkind, 2022-01-24:
The value I provided for `VS_Rgn2K` was already in (Nm/rpm)^2, so no need for the extra unit conversion.

@nikhar-abbas, do you know why [this](https://github.com/NREL/ROSCO/blob/9646dbdab604471477c40c33023ec776a9ac9ec0/Test_Cases/IEA-15-240-RWT-UMaineSemi/linearizations/lin_00.1.lin#L263) has such a low rotor speed?

I suspect the real reason for the mismatch here and there is that the torque controller used in linearizations versus ROSCO is different.  In ROSCO, the generator torque will control the rotor speed so that it's 5 rpm in steady-state.  In the linearization, a \tau = k \omega^2 law is used, so to raise the rotor speed, you'll need to decrease k (`VS_Rgn2K`) to achieve 5 rpm, which might be an iterative process.  In practice, the linear model probably won't change that much from 3.3 to 5 rpm at 5 m/s, but you can try this method to find out.

nikhar-abbas, 2022-01-25:
@dzalkind, I think you're spot on as to why the rotor speed is low here. Those linearizations were done just using the standard servodyn input for `VS_Rgn2K`, so they don't account for the minimum rotor speed of the model and the expected pitched blades. This should probably be updated in ROSCO, but I do agree that the linear model probably won't change significantly.

QusayHawari, 2022-01-25:
Hi,

Thank you for your help.

I changed VS_Rgn2K to 115000 Nm/(rpm^2) to increase the rotor speed as you suggested, and it worked. I got the following OPs.

[image]

I checked both linearized models by plotting the open-loop frequency response from the collective pitch command to the generator speed, and I got the following.

[image]

Note: Blue response is the one at 5rpm OP, while the orange is the one at 3.5 rpm OP.
It can be seen that there is a difference in both open-loop responses, particularly in the range of 0.1-1 rad/sec.

The blue response seems to dip in magnitude at around 0.52rad/s which is the 1p frequency at 5rpm.

Regards,
Qusay.

closed 2022-02-25 by gbarter