IEAWindSystems/IEA-15-240-RWT #79: [Bug] - Running Monopile case with OpenFast 3.0.0

edaniele-tpi, 2022-01-27:
I am using the Windows binary 3.0.0 version of OpenFAST from [here](https://github.com/OpenFAST/openfast/releases/download/v3.0.0/openfast_x64.exe).
While trying to run it with the [Monopile case](https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/OpenFAST/IEA-15-240-RWT-Monopile) I encountered the following error, I presume, while reading the input of `Aerodyn`:

~~~~bash
**************************************************************************************************
OpenFAST

Copyright (C) 2021 National Renewable Energy Laboratory
Copyright (C) 2021 Envision Energy USA LTD

This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
See the "LICENSE" file distributed with this software for details.
**************************************************************************************************

OpenFAST-v3.0.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 1900
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jun 23 2021
 - Time: 21:25:05
Execution Info:
 - Date: 01/27/2022
 - Time: 14:59:26+0100

OpenFAST input file heading:
    IEA 15 MW offshore reference model monopile configuration

Running ElastoDyn.
Nodal outputs section of ElastoDyn input file not found or improperly formatted.
Running AeroDyn.

FAST_InitializeAll:AD_Init:ParsePrimaryFileInfo:ParseSiAry:
 >> A fatal error occurred when parsing data from
 ".\../IEA-15-240-RWT/IEA-15-240-RWT_AeroDyn15.dat".
 >> The "Properties for tower node 1." array was not assigned valid REAL values on line #107.
 >> The text being parsed was :
    "15.           10.             1."

 OpenFAST encountered an error during module initialization.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.
~~~~

Is there any adjustment I should be aware of prior to smoothly running this case?
Thanks in advance for your time and support.

gbarter, 2022-01-27:
For OpenFAST v3.0, please use our `develop branch` (will be moved to `master` shortly):
https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/openfast_dev

edaniele-tpi, 2022-01-27:
By checkout the mentioned `openfast_dev` branch it encounters a different error at runtime:

~~~~bash

**************************************************************************************************
OpenFAST

Copyright (C) 2021 National Renewable Energy Laboratory
Copyright (C) 2021 Envision Energy USA LTD

This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
See the "LICENSE" file distributed with this software for details.
**************************************************************************************************

OpenFAST-v3.0.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 1900
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jun 23 2021
 - Time: 21:25:05
Execution Info:
 - Date: 01/27/2022
 - Time: 18:47:50+0100

OpenFAST input file heading:
    IEA 15 MW offshore reference model monopile configuration

Running ElastoDyn.
Running AeroDyn.
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 5, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 5, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 3)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 3)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 3)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 5, blade 3)
Running InflowWind.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows, ).

FAST_InitializeAll:SrvD_Init:BladedInterface_Init:The dynamic library
.\../IEA-15-240-RWT/ServoData/libdiscon.dylib could not be loaded. Check that the file exists in
the specified location and that it is compiled for 64-bit applications.

 OpenFAST encountered an error during module initialization.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.
~~~~

I do verified that the file `.\../IEA-15-240-RWT/ServoData/libdiscon.dylib` is there where it should, but I have no clue if it is compiled for 64-bit applications. How can I ensure that?

gbarter, 2022-01-27:
You need to update the ServoDyn file to point to the controller DLL which you should have from this folder:
https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/openfast_dev/OpenFAST/IEA-15-240-RWT/ServoData

The `dylib` extension is for Mac.  `dll` is for Windows.

edaniele-tpi, 2022-01-27:
Fair enough, `dll` is now rightly called.
Though I faced an issue with `HydroDyn`:

~~~~bash

**************************************************************************************************
OpenFAST

Copyright (C) 2021 National Renewable Energy Laboratory
Copyright (C) 2021 Envision Energy USA LTD

This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
See the "LICENSE" file distributed with this software for details.
**************************************************************************************************

OpenFAST-v3.0.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 1900
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jun 23 2021
 - Time: 21:25:05
Execution Info:
 - Date: 01/27/2022
 - Time: 19:07:07+0100

OpenFAST input file heading:
    IEA 15 MW offshore reference model monopile configuration

Running ElastoDyn.
Running AeroDyn.
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 5, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 5, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 3)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 3)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 3)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 5, blade 3)
Running InflowWind.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows, ).
Using legacy Bladed DLL interface.
Running HydroDyn.

FAST_InitializeAll:HydroDyn_Init:HydroDynInput_GetInput:Failed to read joints.

 OpenFAST encountered an error during module initialization.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.
~~~~

Any input file adjustment needed here?

gbarter, 2022-01-28:
Not sure how those errors crept in, but they are fixed now.  If you do a `git pull`, things should run with OpenFAST v3.0. Thanks for catching that bug!

edaniele-tpi, 2022-01-28:
Thanks again, everything look fine now.

~~~~bash
**************************************************************************************************
OpenFAST

Copyright (C) 2021 National Renewable Energy Laboratory
Copyright (C) 2021 Envision Energy USA LTD

This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
See the "LICENSE" file distributed with this software for details.
**************************************************************************************************

OpenFAST-v3.0.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 1900
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jun 23 2021
 - Time: 21:25:05
Execution Info:
 - Date: 01/28/2022
 - Time: 09:06:20+0100

OpenFAST input file heading:
    IEA 15 MW offshore reference model monopile configuration

Running ElastoDyn.
Running AeroDyn.
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 5, blade 1)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 5, blade 2)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 3)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 3)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 4, blade 3)
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 5, blade 3)
Running InflowWind.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows, ).
Using legacy Bladed DLL interface.
Running HydroDyn.
 Generating incident wave kinematics and current time history.
Running SubDyn.
   Fixed bottom case detected
   Performing Craig-Bampton reduction 60 DOFs -> 0 modes + 6 DOFs
   Using static improvement method for gravity and ext. loads
   Calculating Full System Modes for summary file
Generator speed:    6.8 RPM, Pitch angle:   0.0 deg, Power: 12116.6 kW, Est. wind Speed:  10.1 m/s
Generator speed:    7.0 RPM, Pitch angle:   0.0 deg, Power: 11883.8 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.1 RPM, Pitch angle:   0.0 deg, Power: 12197.3 kW, Est. wind Speed:  10.1 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12402.3 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12435.2 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12419.1 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12425.3 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12450.5 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12459.7 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12416.8 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12445.1 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12464.6 kW, Est. wind Speed:  10.1 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12459.0 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12393.8 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12457.6 kW, Est. wind Speed:  10.1 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12475.4 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12453.0 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12430.9 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12468.3 kW, Est. wind Speed:  10.1 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12477.7 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12418.5 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12431.9 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12472.1 kW, Est. wind Speed:  10.1 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12448.6 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12434.5 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12446.8 kW, Est. wind Speed:  10.1 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12464.9 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12434.8 kW, Est. wind Speed:  10.0 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12420.9 kW, Est. wind Speed:  10.1 m/s
Generator speed:    7.2 RPM, Pitch angle:   0.0 deg, Power: 12464.7 kW, Est. wind Speed:  10.1 m/s

 Total Real Time:       15.116 minutes
 Total CPU Time:        14.826 minutes
 Simulation CPU Time:   14.798 minutes
 Simulated Time:        5 minutes
 Time Ratio (Sim/CPU):  0.33789

 OpenFAST terminated normally.
~~~~

closed 2022-01-28 by edaniele-tpi