IEAWindSystems/IEA-15-240-RWT #87: FEM model (shell7.src)

isadora-toledo, 2022-04-11:
I am a master's student investigating the applicability of shell models in wind turbine blade structural analysis. I have read in the FAQs that you have withdrawn from the repo the high-fidelity FEM model (shell7.src) and will bring it back after extensive testing. 

I am working on that model, mainly comparing the results of static and modal analyses with a nonlinear beam model (BeamDyn/also available in the repo). I would like to point out that, although my research is not yet complete, the global behavior from the shell model seems to be compatible with the beam.

Based on this feedback, I ask you to bring the shell7.src model back to the repo because it may be handy for other people. I can give you more feedback later when the conclusions are more mature if you are interested.

Best regards,

Isadora

gbarter, 2022-04-11:
One advantage of using version control for the reference models is that the prior history is never lost.  You can find the v1.0 files here:
https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/v1.0/CAD/IEA-15-240-RWT_Ansys

isadora-toledo, 2022-04-11:
Thank you very much, @gbarter !

closed 2022-04-11 by isadora-toledo