IEAWindSystems/IEA-15-240-RWT #88: Request for 64 bit version of libdiscon.so OpenFAST v 3.1.0 UMaine VolturnUS-S Semisub

andreowhite, 2022-04-19:
Hi @gbarter I have started using OpenFAST v3.1.0 and I am getting an error when I try to run a simulation for the VolturnUS-S Semisub. The error is shown below:
**FAST_InitializeAll:SrvD_Init:BladedInterface_Init:The dynamic library** 
**.\../IEA-15-240-RWT/ServoData/libdiscon.so could not be loaded. Check that the file exists in the** 
**specified location and that it is compiled for 64-bit applications.** 
 
** OpenFAST encountered an error during module initialization.** 
 **Simulation error level: FATAL ERROR**
 **Aborting OpenFAST.** 

I believe that this error may be because I am using the 32 bit version of libdiscon.so. Can you please let me know where I can get the 64 version of libdiscon.so for OpenFAST v3.1.0. Otherwise if it is another error do you have a fix for it?

Thank you.

Regards,
AOAW

ptrbortolotti, 2022-04-19:
Hello AOAW,
please go ahead and compile ROSCO (https://github.com/NREL/ROSCO) yourself on your machine(s) using the instructions available here https://rosco.readthedocs.io/en/latest/source/install.html
Please let us know if this step isn't successful.
Best regards,
Pietro

gbarter, 2022-04-19:
I would also suggest making sure you are current with the latest release (v1.1.2).  The ROSCO team recently updated their [files](https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/OpenFAST/IEA-15-240-RWT/ServoData) and there are 64-bit libraries for all platforms, and 32-bit for Windows.

andreowhite, 2022-04-19:
Thanks @gbarter and @ptrbortolotti the crux of the problem was that I was using the .so file which is for Linux. I needed to have used the .dll file which is for Windows. 
I think when I had tried to run in OpenFAST vs2.5.0 - some time ago -  I had the same issue. Lol totally forgot that initially.  I eventually remembered and was able to retrace to how the controller was defined in the servodyn. Once I changed to .dll that solved it.

Thanks again.

Regards,
AOAW

closed 2022-06-18 by gbarter