IEAWindSystems/IEA-15-240-RWT #90: Error in OpenFAST running UMaine VolturnUS-S floater for the wave only condition

andreowhite, 2022-06-18:
Hi everyone:
I ran the VolturnUS-S floater for the wave only condition without any problems on numerous occassions. I set up the wave only case by setting InflowWind and AeroDyn to 0 in the .fst file. However, I just attempted to run the same case and I received the following error in OpenFAST:
**FAST_Solution:FAST_AdvanceStates:ED_ABM4:ED_CalcContStateDeriv:SetCoordSy:Small angle assumption
violated in SUBROUTINE SmllRotTrans() due to a large blade deflection (ElastoDyn SetCoordSy). The
solution may be inaccurate. Simulation continuing, but future warnings from SmllRotTrans() will
be suppressed.**
 **Additional debugging message from SUBROUTINE SmllRotTrans(): 2.725 s
Generator speed:    8.7 RPM, Pitch angle:  13.2 deg, Power: 19196.6 kW, Est. wind Speed:   0.0 m/s
 Time: 10 of 1800 seconds.  Estimated final completion at 06:47:39.**

**FAST_Solution:CalcOutputs_And_SolveForInputs:SolveOption1:ED_HD_InputOutputSolve:HydroDyn_CalcOutp
ut: Angles in GetSmllRotAngs() are larger than 0.4 radians.**
**HydroDyn_CalcOutput:HDOut_MapOutputs: Angles in GetSmllRotAngs() are larger than 0.4 radians.
ED_HD_InputOutputSolve:HydroDyn_CalcOutput: Angles in GetSmllRotAngs() are larger than 0.4
radians.**
**HydroDyn_CalcOutput:HDOut_MapOutputs: Angles in GetSmllRotAngs() are larger than 0.4 radians.**

I am not sure what the problem is as it is reporting large blade deflection even though the wind is turned off and there seems to be an error with HydroDyn. I am attaching the ElastoDyn and HydroDyn input files to see if someone could help me to decipher where the error is located.
[IEA-15-240-RWT-UMaineSemi_ElastoDyn2.txt](https://github.com/IEAWindTask37/IEA-15-240-RWT/files/8933052/IEA-15-240-RWT-UMaineSemi_ElastoDyn2.txt)
[IEA-15-240-RWT-UMaineSemi_HydroDyn100m_9mpsJCS1.txt](https://github.com/IEAWindTask37/IEA-15-240-RWT/files/8933062/IEA-15-240-RWT-UMaineSemi_HydroDyn100m_9mpsJCS1.txt)

Thank you.

Regards,
AOAW

mattEhall, 2022-07-12:
Hi Andre, this would probably be a good question for the NREL Forum if you haven't already asked it there. Probably the first thing to do would be to set the rotor speed to zero and turn off the generator DOF as well as control. But I bet you've already solved it at this point, and in any case the forum would be the place where you'd get the best answers.

Cheers,
Matt

gbarter, 2022-07-12:
I will close this because there is an active dialogue in the forums.

closed 2022-07-12 by gbarter

andreowhite, 2022-07-12:
Ok Garrett.

Regards,
AOAW