IEAWindSystems/IEA-15-240-RWT #91: UMaine VolturnUS-S floater - WAMIT parameters for wave directions other than zero

andreowhite, 2022-06-18:
Hi everyone:
I would like know if anyone has determined the WAMIT parameters (hydrodynamic added mass and damping coefficients etc.) for wave directions other than 0 for the VolturnUS-S floater that was presented by Allen et al (2020). That is 30, 60, 90 etc wave directions or any other direction greater than 0. If anyone has done it, I would be very grateful if you could sare them with me. I would like to assess the FOWT system for wave directions other than zero but I am not able to do so as I not know how to use WAMIT as yet. I am not averse to learning it but due to severe time constraints, it is not possible now.

Thank you.

Regards,
AOAW

gbarter, 2022-06-21:
Hi Andre,

I believe we only ran the 0-deg case and did the design around that.  The input files used are [here](https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/HydroData/wamit_inputs):

@mattEhall - If you have other directions calculated, please chime in.

Cheers,
Garrett

andreowhite, 2022-06-21:
Thanks and noted @gbarter .

Regards,
AOAW

andreowhite, 2022-07-12:
Noted Matt. That sounds good. I would really appreciate it. When you have the outputs for these wave headings please send to me.

Thank you.

Regards,
AOAW

closed 2024-07-11 by gbarter