IEAWindSystems/IEA-15-240-RWT #92: .OUT WAMIT file for implementation in Bladed

malteoliver, 2022-06-20:
Hi all,

In order to run a simulation of the Volturn US using the software "Bladed" from DNV, I would like to upload the WAMIT files to the simulation program. As far as I understand it, Bladed requires the follwoing WAMIT files with these endings:
- .out
- .pot
- .hst
- .1
- .2 (or .3)
- .cfg
- .frc
I was able to find all files except the **.out** file. Does anyone know what the file is called or where I can find it?

I would be very happy if someone can help me out with my problem. Thanks in advance! :)

gbarter, 2022-06-21:
@mattEhall - Would you happen to have the .out file from the WAMIT runs?

mattEhall, 2022-07-11:
Hi @malteoliver, sorry for my delay following up on this. I thought I would have to dig up the WAMIT files I used for the runs, but once I finally did that and prepared to add the .out files, I realized they are already on the repo! 

You'll find them here: https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/develop/OpenFAST/IEA-15-240-RWT-UMaineSemi/HydroData  They have a modified externsion - ".out.wamit" - probably to avoid a filter that ignores .out files. If you need anything else just let me know. I have everything close at hand now.

mattEhall, 2022-07-12:
@malteoliver, one more comment on this:
As you probably realized, we used separate WAMIT runs to generate the first-order and second-order WAMIT data. However, we only provided one set of input files on the repo. I have now updated my repo fork to include the input files for both WAMIT runs. They are in separate folders, and I moved the .out files into the respective folders as well. You can find them here for now: https://github.com/mattEhall/IEA-15-240-RWT/tree/develop

@gbarter, here is a pull request for this update (which I should have done many months ago): https://github.com/IEAWindTask37/IEA-15-240-RWT/pull/95