IEAWindSystems/IEA-15-240-RWT #94: Modelling Parked (Standstill) condition for VolturnUS-S Floating Wind Turbine

andreowhite, 2022-07-08:
Hi everyone:

I am trying to model the parked/standstill condition in openfast using VolturnUS-S semisubmersible. I have set GenDOF = False, feathered blades to 90 degrees and set initial rotor speed to zero in ElastoDyn. PCMode=0 in ServoDyn.

In AeroDyn, I have set WakeMoD=0 and AFAeroMod=1.0.

However, I get this error:

FAST_Solution:FAST_AdvanceStates:ED_ABM4:ED_CalcContStateDeriv:SetCoordSy:Small angle assumption

violated in SUBROUTINE SmllRotTrans() due to a large blade deflection (ElastoDyn SetCoordSy). The

solution may be inaccurate. Simulation continuing, but future warnings from SmllRotTrans() will

be suppressed.

Additional debugging message from SUBROUTINE SmllRotTrans(): 0.725 s

FAST_Solution:FAST_AdvanceStates:SolveOption2c_Inp2AD_SrvD:InflowWind_CalcOutput:CalcOutput:IfW_FF

Wind_CalcOutput [position=(5.5972, -4.9906, 1.863) in wind-file coordinates]: FF wind array

boundaries violated. Grid too small in Z direction (height (Z=1.863 m) is below the grid and no

tower points are defined).

OpenFAST encountered an error at simulation time 0.95 of 3600 seconds.

Simulation error level: FATAL ERROR

In the z-direction my grid goes from 2.5 to 297.5 m that is grid height is 295 m. This is the limit based on grid height criteria in TurbSim.

Do you know how I could fix this problem?

Regards,
AOAW

ptrbortolotti, 2022-11-06:
Hello AOAW,
this is a very late reply, my apologies for it, I've missed this issue.
The small angle assumption being violated is a warning, not an error. The violation likely occurs because of some unrealistic transient happening at the beginning of the simulation, I wouldn't worry too much about it. The real error is the second one and it is likely caused by the tower aerodynamic nodes being out of your wind grid. Unfortunately TurbSim does not allow the wind grid to start from the ground, and therefore we usually start it at a finite height, say 1m. This however means that when OpenFAST models the inflow with TurbSim (wind type 3 in InflowWind), you need to set the tower aero nodes above the ground.
Does this help?
Best regards,
Pietro

andreowhite, 2022-11-09:
Thanks Pierto.
I think I had eventually figured this out.

Thanks again.
Regards,
AOAW

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

closed 2022-11-09 by gbarter