IEAWindSystems/IEA-15-240-RWT #96: Incomplete elastic_properties_mb?

pelljam, 2022-08-15:
In _WT_Ontology/IEA-15-240-RWT.yaml_ , are the stiffness matrices under _elastic_properties_mb_ correct? Possibly I've misinterpreted them, but they look very sparse compared to e.g. the BeamDyn data set?

Thx, James

ptrbortolotti, 2022-08-22:
Hello James, 
my apologies for the slow reply. You are correct, we have an inconsistency there. The properties in the yaml are sparse because they were generated within WISDEM using PreComp, which ignores shear stiffness and several others off-diagonal terms. The BeamDyn files were generated using SONATA and ANBA4 (https://github.com/ptrbortolotti/SONATA/blob/master/examples/1_IEA15MW/1_sonata_IEA15.py). I will make sure to import the latter props into the yaml file.
Best,
Pietro

pelljam, 2022-08-23:
Great, thank you.

closed 2022-09-21 by gbarter