IEAWindSystems/IEA-15-240-RWT #97: Ontology file for UMaineSemi tower

davidovitch, 2022-08-16:
Is it possible that there is no ontology `yaml` description yet of the UMaineSemi tower? I can see there is `OpenFAST/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_ElastoDyn_tower.dat` for the OpenFAST model so I can of course also use that. I am currently converting an earlier version of the HAWC2 model with the UMaineSemi floater to fit/align with the recently re-organised/updated HAWC2 model.

gbarter, 2022-08-16:
Are you looking for this [line](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/WT_Ontology/IEA-15-240-RWT_VolturnUS-S.yaml#L556) in the ontology?

davidovitch, 2022-08-17:
It seems I didn't realise that `UMaineSemi` and `VolturnUS-S` are different names for the same floater. I have also seen another tower definition being used as the one linked from the ontology but I am not sure where that came from.

gbarter, 2022-08-17:
Good point on the naming inconsistency.  Certainly, if you find inconsistencies in the design definitions between the model files and ontology, please let us know!

davidovitch, 2022-08-17:
I will close this issue and open another in case I find track down the origin of the other tower, and/or with the HAWC2 model of this floater/tower.

davidovitch, 2022-08-17:
The reference report of the UMaine floating tower looks different than what is in the ontology. Has there been a re-design since the report was published, and is that what is in ontology `yaml` file? At least that seems what is going in this commit 1a1cbf57aeee5eff1e4ee99f0aafb16f4333fbc6. I also remember from @jennirinker the original tower+floater had an eigenfrequency close to 3P and had to be stiffened.

gbarter, 2022-08-17:
@davidovitch : Exactly.  The original tower described in the report did not achieve the original design intent.  It was modified in v1.1 (see release [page](https://github.com/IEAWindTask37/IEA-15-240-RWT/releases) and [notes](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/ReleaseNotes.md)).

davidovitch, 2022-08-18:
Thanks a lot for clarifying @gbarter

closed 2022-08-18 by davidovitch