IEAWindSystems/IEA-15-240-RWT #103: Request for updated WindCrete floater of the 15MW model

QusayHawari, 2022-10-18:
Hi, 

I wanted to ask if I can find updated inputs files for the 15MW WindCrete floater. 

The only one I found was in the link below, which is not up to date with the latest OpenFAST and ROSCO.

https://zenodo.org/record/4322446#.Y069b6DMKUk

Thanks in advance,
Qusay

gbarter, 2022-10-18:
Yes, you would have to change all of those, and the SubDyn file, to switch to a spar.  More info on appropriate modeling of spars in OpenFAST is available on the NREL Forums:
https://forums.nrel.gov/

QusayHawari, 2022-10-18:
Thanks for your reply.

I am using a floating model not a fixed bottom structure. No need for SubDyn I think?

Qusay.

gbarter, 2022-10-18:
That depends on what you are interested in.  Newer versions of OpenFAST compute member-level loads of floaters via SubDy, but it isn't a required element.

closed 2022-10-31 by gbarter