IEAWindSystems/IEA-15-240-RWT #106: Blade Pitch Behavior of the IEA-15-240-RWT-Monopile openFAST model

fnaos, 2022-11-06:
Hello,

I recently simulated a staircase wind scenario utilizing the IEA-15-240-RWT-Monopile openFAST model with the ROSCO controller. When comparing the openFAST results with the rotor performance values found in Documentation/IEA-15-240-RWT_tabular.xlsx, I noticed some differences and I have been wondering what could be the potential cause for that discrepancies. 

In the picture below you can visualize that the simulated blade pitch of the turbine does not follow the reference values found in the excel sheet with a large deviation around the transition regions.  Any idea what might be leading to that behavior?

I haven't changed any controller parameters and I am using openFAST v3.1.0 for my simulation (performed the regression test).

[image]

Best,
Felipe.

gbarter, 2022-11-06:
Hi Filipe- The current tabular data is populated by WISDEM, which is a lower-fidelity, steady-state model without full aero-elastic coupling.  The pitch schedule can be thought of as a an idealized, and simplified, approximation of what would be seen in an OpenFAST simulation with full coupling and an active controller.  I would fully expect the differences to be most pronounced in the transition region as ROSCO switches control paradigms, exactly as you have seen.

ptrbortolotti, 2023-01-13:
Hi Felipe, I will close this now, please open a new issue if you still need help

closed 2023-01-13 by ptrbortolotti