IEAWindSystems/IEA-15-240-RWT #109: Stability of v1.1.2

pelljam, 2022-11-21:
Hi all, 

I’ve been investigating the v1.1.2 turbine’s step-wind response using OpenFAST and an equivalent OrcaFlex model (this study is very similar to that of figure 4 in [Comparison of loads from HAWC2 and OpenFAST for the IEA Wind 15 MW Reference Wind Turbine - IOPscience](https://iopscience.iop.org/article/10.1088/1742-6596/1618/5/052052)).

I’m working with the UMaineSemi variant, however I have disabled the floater’s motions. So it is essentially a tower only model. The only other changes I have made are to disable the floating feedback for the ROSCO controller and to update the AeroDyn tower model to reflect the UMaineSemi tower (FYI as distributed, this always represents Monopile tower).

As per the above paper, the step wind ranges from 3 to 25 m/s, and each step is 40 s. Once the wind speed gets up to >20 m/s, I’m seeing significant blade oscillations. Here are some key results (OpenFAST in red):

[image]

[image]

I just wondered if you had experienced this, had any plans to modify the model to avoid it and, if so, had any advice on how we might modify the model now?

FWIW, I notice things are exacerbated (to the point of not being able to run the simulation) by turning off the unsteady aerodynamics in AeroDyn, and if I change to use the Gonzalez variant, then things are greatly improved. Maybe the load hysteresis those models introduce are disrupting a flutter instability (assuming that is the issue)?

Thanks for your help,

James

jjonkman, 2022-11-26:
Dear @pelljam,

I don't see that the BeamDyn model of the IEA Wind 15-MW reference wind turbine has been shared through this repository.  Can you clarify the source of the BeamDyn model you are using?

Regardless, I agree that--given that the unsteady airfoil aerodynamics model, which will impact the blade aerodynamic damping, has a strong influence on the response--that you are likely experiencing negative blade damping, e.g., a flutter instability, with your BeamDyn model.

Best regards,

pelljam, 2022-11-28:
Dear @jjonkman, 

Thank you for your response; we really appreciate your advice and the work on these RWT.

With regards to BeamDyn. I’m just working from the files available through the v1.1.3 release. This appears to contain a fully populated BeamDyn dataset (as do the previous releases, as far as I am aware)? The FAQ, suggests this is intentional:

_Can I use BeamDyn?
The OpenFAST model files are configured to use ElastoDyn by default. To use BeamDyn, set DT = 0.0005 and CompElast = 2 in the FST-file._

That’s the process I have followed. I know it considers an earlier revision of the RWT, but the paper I linked above also suggested that the BeamDyn data set was valid. Have I misinterpreted this? 

The ElastoDyn model naturally won’t be susceptible to these kinds of blade dynamics. This is why I wondered if you had much experience of these oscillations in BeamDyn, i.e if a lot of the work has been done with ElastoDyn then perhaps it’s not presented itself as much of an issue. I’ve not used the HAWC2 models, but I assume they too might display this behaviour?

I’d be interested to know if you think working from the BeamDyn (or, I guess, equivalently HAWC2 data set) is the best approach? Or if these torsional blade dynamics are possibly unrealistic, because the RWT design has not focused on them yet, and it is better to suppress them through the choice of OpenFAST structural solver (or artificially high torsional stiffness in an equivalent OrcaFlex model).

Best regards,

James

ptrbortolotti, 2022-11-28:
Hello James,
This is all very interesting, thank you for sharing your findings.
This is an active area of research at NREL (and elsewhere, see for example https://iopscience.iop.org/article/10.1088/1742-6596/1618/5/052048). We have a publication pending investigating the aeroelastic stability of various designs using BeamDyn. We hope to submit it soon. In the meanwhile I can confirm that the linearized model of the IEA15 shows an instability above 20 m/s. We did not investigate it thoroughly since we're focusing on very flexible rotors from the BAR project (https://wes.copernicus.org/articles/6/1277/2021/) and the RAAW project (https://www.nrel.gov/wind/assets/pdfs/engineering-wkshp2022-6-1-naughton.pdf). And you are right, we did not account for instabilities during the design phase. We will for the new offshore reference wind turbine that will come online in the next weeks/months. I can finally confirm that we also see a large impact of the aerodynamic models on those instabilities and we're working on validating the numerical predictions.
Please let me know if you have further questions.
Thank you and best regards,
Pietro

pelljam, 2022-11-29:
Hi @ptrbortolotti, thank you for your reply and pointing me to all these resources. That's really useful. We look forward to the new RWT.

Best Regards, James

closed 2022-11-29 by pelljam