IEAWindSystems/IEA-15-240-RWT #110: Who has the.DLL file for IEA-15-240-rwt\ServoData

lqw075, 2022-12-05:
Dear all,

The comment on line 77 of the ServoDyn input file for the IEA-15-240-RWT-UMaineSemi_ServoDyn.dat indicates that a .dll file should be used for Windows, while only a .so (Linux) is available in the IEA-15-240-RWT.  Would it be possible to make the .dll file available to Windows users?

Thank your for your attention.
Best regards,

gbarter, 2022-12-05:
Compiled ROSCO libraries for all operating systems are available here:
https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/OpenFAST/IEA-15-240-RWT/ServoData

lqw075, 2022-12-05:
> Compiled ROSCO libraries for all operating systems are available here: https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/OpenFAST/IEA-15-240-RWT/ServoData

The problem has been solved.Thank you very much!

closed 2023-01-13 by ptrbortolotti