IEAWindSystems/IEA-15-240-RWT #112: BECAS input file for fatigue analysis of blades

MassimoSirigu, 2022-12-12:
Hello,
I would ask you if there is a repository to find the input files for BECAS (regarding the section properties of the blade of the IEA 15 MW wind turbine).
Thank you for your attention
Best,
Massimo Sirigu

ptrbortolotti, 2022-12-12:
Hello Massimo,
an alternative to BECAS is the combination SONATA/ANBA4, which are harder to use but 100% open-source. Example #1 is about the IEA15 https://github.com/ptrbortolotti/SONATA/blob/master/examples/1_IEA15MW/1_sonata_IEA15.py
Best regards,
Pietro

MassimoSirigu, 2022-12-12:
Hello Pietro,
I tried to use SONATA but at the moment I was not able to use it in the virtual machine (for Linux). For this reason, I tried BECAS and since this software is implemented in Matlab I found it much more useful for us, especially because it is well-documented also for failure criteria and fatigue.
So I'm trying to realise the sections' properties to use them in BECAS.
So I would ask you: in the YAML file, how are the properties defined? I'm struggling to understand some terms:
for the webs, what does it mean in YAML file:
rotation, offset_y_pa, start_nd_arc and end_nd_arc
In the becas python code I have to define the starting node and the ending node of the web, referring to the span of the airfoil, and also the material and thickness of the composites of the airfoil.

If you want, If I manage to define the profiles for BECAS, do you think they can be useful for you? or uploaded here  for anyone else?
 at the moment it gives me this result:
[image]

jennirinker, 2022-12-13:
@fzahle Do you know of anyone who has the BECAS files lying around? I never used them, myself.

closed 2024-07-11 by gbarter