IEAWindSystems/IEA-15-240-RWT #115: Running version 3.1

DanielaMarramiero, 2023-01-17:
Hello,

I have downloaded the latest version of IEA-15-240-RWT, didn't modify anything, only added a link to a working libdiscon.so, and tried running it using version 3.1 of OpenFAST. 
The following error appears: 

 Time: 0 of 300 seconds.
At line 78 of file /home/conda/feedstock_root/build_artifacts/rosco_1615353126851/work/src/ReadSetParameters.f90 (unit = 89, file = './IEA-15-240-RWT-Monopile_DISCON.IN')
Fortran runtime error: Bad real number in item 1 of list input

Error termination. Backtrace:
#0  0x7f7cbe191dfd in ???
#1  0x7f7cbe192995 in ???
#2  0x7f7cbe19317d in ???
#3  0x7f7cbe38d778 in ???
#4  0x7f7cbe38fc6d in ???
#5  0x7f7cbe391219 in ???
#6  0x7f7cbe396721 in ???
#7  0x7f7cc5162284 in ???
#8  0x7f7cc5164624 in ???
#9  0x7f7cc515a290 in ???
#10  0x134e188 in ???
#11  0x134e367 in ???
#12  0x134e5d5 in ???
#13  0x135ca09 in ???
#14  0x1380e8c in ???
#15  0x64d3f3 in ???
#16  0x64d73d in ???
#17  0x5a52b7 in ???
#18  0x5a5713 in ???
#19  0x4ecea7 in ???
#20  0x4042de in ???
#21  0x7f7cbd169554 in ???
#22  0x40430e in ???
#23  0xffffffffffffffff in ???

Do you have any suggestions to solve this? 

Thank you for your time. 

Best regards,
Daniela

dzalkind, 2023-01-17:
Hi Daniela,

What version of ROSCO are you using with which DISCON input?  The DISCONs available on this repo are set up to run with [ROSCO 2.6.0](https://github.com/NREL/ROSCO/releases/tag/v2.6.0).

Best, Dan

gbarter, 2023-01-17:
I think the current input files are consistent with OpenFAST v3.3.  OpenFAST released v3.4 last week and we haven't yet made the input file modifications for that yet.

DanielaMarramiero, 2023-01-17:
I am using the DISCON file that is in the Monopile folder. About ROSCO, I downloaded version 2.6 and tried using those files but I get this error.

 ./../IEA-15-240-RWT/ServoData/libdiscon.so could not be loaded. Check that the file exists in the
 specified location and that it is compiled for 64-bit applications.

The file is indeed in the correct folder and I am just using the files downloaded from the link Dan sent me. 

 What am I doing wrong?

dzalkind, 2023-01-17:
This error also appears when ROSCO and OpenFAST have been compiled with different compilers.  You may need to re-compile ROSCO 2.6.0 from source with the same compilers you used for OpenFAST.

https://rosco.readthedocs.io/en/latest/source/install.html#compile-using-cmake

DanielaMarramiero, 2023-01-17:
Thank you for your replies. I followed your suggestion but I still get the same error. Any other ideas?

gbarter, 2023-01-17:
Can you share a bit more about your environment?  Are you on a Windows or Mac or Linux?  Did you successfully switch to OpenFAST v3.3?  Did you modify the ServoDyn library entry to match your machine (.so vs .dll)?

DanielaMarramiero, 2023-01-20:
Sorry for the late reply and thank you for your comment. I am on Linux and was running OpenFAST v3.1. I think that I made a mistake in the compiling of the files. Thank you all for your help!

closed 2023-01-20 by DanielaMarramiero