IEAWindSystems/IEA-15-240-RWT #120: IEA 15MW Turbine - OpenFAST error

mocominnie, 2023-02-08:
Hi, 

I'm trying to run the 15MW reference turbine (on ubuntu) but encounter the error below. I'm not entirely sure what's happening as I followed the install instructions in the docs. For some reason it seems to work ok my mac.

Thanks for your help.

OpenFAST input file heading:
IEA 15 MW offshore reference model on UMaine VolturnUS-S semi-submersible floating platform

 Running ElastoDyn.
 Nodal outputs section of ElastoDyn input file not found or improperly formatted.
 Running AeroDyn.

 FAST_InitializeAll:AD_Init:ParsePrimaryFileInfo:ParseLoVar:
  >> A fatal error occurred when parsing data from "./IEA-15-240-RWT_AeroDyn15.dat".
  >> The variable "Buoyancy" was not found on line #13.

 OpenFAST encountered an error during module initialization.
  Simulation error level: FATAL ERROR

  Aborting OpenFAST.

gbarter, 2023-02-08:
This is just an issue with OpenFAST changing their input file formatting in the recently released v3.4.  The files in the `develop` branch are consistent with the latest release.  In the short-term, I would recommend using the files in that branch.  We will hopefully bring those changes into the `master` branch soon.

mocominnie, 2023-02-09:
Thanks for your help, it's working now.

closed 2023-02-09 by mocominnie