IEAWindSystems/IEA-15-240-RWT #121: #Request the geometric characteristics of the blade cross-section#

li-Bjut, 2023-02-15:
Hello,
I am a graduate student, and I am currently working on projects related to offshore wind turbines. I would like to build a complete IEA-15MW model in the ABAQUS software, requiring the section bending stiffness and torsional stiffness of the blade and the section area. In the Openfast input file(IEA-15-240-RWT_ElastoDyn_blade.dat) I found BMassDen, FlpStff, EdgStff. But EAStff and GJStff I can t find. The curve of torsional stiffness is mentioned in the IEA 15MW definition file, and I tried to take the value from the curve. But clearly the results are not accurate. 
So I would like to know if it is convenient for you to provide the blade of EAStff and GJStff.
Thanks.
Regards,
Anna

gbarter, 2023-02-15:
You can find the full 6x6 mass & stiffness matrices for the blade in the OpenFAST [BeamDyn file](https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_BeamDyn_blade.dat)

li-Bjut, 2023-02-15:
Thank yuo for your reply.
I have just browsed this file and found the mass and stiffness matrices of 26 sections. I think the area of the blade section I have been able to solve. 
But the torsional stiffness of the blade I still don't get to. 
Any reply from you would be greatly appreciated.
Thanks,
Anna

gbarter, 2023-02-15:
@ptrbortolotti can hopefully comment on the torsional stiffness

ptrbortolotti, 2023-02-16:
Hello Anna,
ElastoDyn does not have the torsional degree of freedom, you need to extract it from either the BeamDyn file https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/OpenFAST/IEA-15-240-RWT/IEA-15-240-RWT_BeamDyn_blade.dat or this excel https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/master/Documentation/IEA-15-240-RWT_tabular.xlsx, tab Blade Structural Properties. Torsional stiffness is K66 (along z axis)
Best regards,
Pietro

ptrbortolotti, 2023-02-16:
K66 is also available here https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/3a00f7f38a6373f6b026aef5878b671ca7af3605/WT_Ontology/IEA-15-240-RWT.yaml#L334. It's the last element of each array

li-Bjut, 2023-02-16:
Thank you so much for your reply, and my question was solved.
Best regards,
Anna

closed 2023-02-16 by li-Bjut