IEAWindSystems/IEA-15-240-RWT #122: possible errors in blade structure details in YAML file

MassimoSirigu, 2023-03-07:
Hello, 
I want to report some possible errors in the blade structure definition in the YAML file.
In particular, it seems that the layers named "shell_skin" and "shell_skin_inner" (triaxial fibreglass) are not present all along the surfaces of the blade, since the length of these layers along the airfoil perimeter is about 1 to 92%.
In this way, it happens that at the tip of the blade, in the trailing edge region, there is only a thin layer of gel coat.
I think in a structural analysis it is better to consider these layers continuous, right?
The second problem is that some data are missing, like the transversal and shear stress of the carbon fibre, and it seems also that the biaxial fibreglass has a low longitudinal tensile strength ( 4.29E7 Pa, compared to 3.96E8 Pa of the triaxial glass). I think it should be one order of magnitude greater.

ptrbortolotti, 2023-03-07:
Hello Massimo,
Thanks for your note. Let me try to help.
Shell skin: why do you say that it stops at 92% span? Here https://github.com/IEAWindTask37/IEA-15-240-RWT/blob/3a00f7f38a6373f6b026aef5878b671ca7af3605/WT_Ontology/IEA-15-240-RWT.yaml#L88 thickness is specified all along
Carbon fiber props: this is a known weakness, see wiki https://github.com/IEAWindTask37/IEA-15-240-RWT/wiki/Frequently-Asked-Questions-(FAQ)#are-the-cfrp-properties-wrong
Biaxial glass: this might be a very good catch. Those material properties have an unclear origin. In the new IEA22, we are using new materials https://github.com/IEAWindTask37/IEA-22-280-RWT/blob/7f6eebcb8ea2e7c4cf25748f3aba05c24ba8bd8e/windIO/IEA-22-280-RWT.yaml#L825 
Best regards,
Pietro

MassimoSirigu, 2023-03-07:
I referred to (in yaml file but also in the excel file): 
for shell skin:
start_nd_arc:
                      grid: [0.0, 0.25, 0.3, 1.0]
                      values: [0.0, 0.0, 0.01, 0.01]
                  end_nd_arc:
                      grid: [0.0, 0.25, 0.3, 1.0]
                      values: [1.0, 1.0, 0.99, 0.99]
end shell skin inner: 
            start_nd_arc:
                      grid: [0.0, 0.2, 0.25, 1.0]
                      values: [0.0, 0.0, 0.08, 0.08]
                  end_nd_arc:
                      grid: [0.0, 0.2, 0.25, 1.0]
                      values: [1.0, 1.0, 0.92, 0.92]
the values are not 1.0 everywhere, so it happens that at the tip of the blade (when the uniaxial fibre for the trailing edge has zero thickness) the only layer is the gel coat.
Thank you for the link, I will use those material properties for my next paper about the fatigue analysis of the blade!

ptrbortolotti, 2023-03-07:
hello Massimo,
start_nd_arc and end_nd_arc refer to the chord wise extension of the layer. The span wise extension is defined by the thickness field, which has two arrays: grid and values. The layers don't extend all the way to the trailing edge because they don't physically fit in the sharp trailing edge shapes. We then assumed that the layers stopped before going all the way out, even if a detailed modeling of the trailing edge structure would be needed.
Regards,
pietro

MassimoSirigu, 2023-03-08:
Thank you for the explanation. I also noted this difficulty when modelling the blade in BECAS. Just for curiosity, I will plot the difference between the meshes in SONATA (present here in GitHub)  and my mesh on BECAS:
[image]
[image]
as you can see in the picture, in SONATA it seems that the trailing edge is full, whereas in BECAS it follows the line of airfoil

ptrbortolotti, 2023-03-08:
That's correct, SONATA and BECAS model the TE differently. We've noted quite an impact from those modeling aspects, but we haven't had the chance to implement different TE modeling options

MassimoSirigu, 2023-03-08:
Thank you for your comment. I would like to share with you also the resulting stiffness with BECAS.
as you can see, the K33 and K44 are a little bit different. I think that since these stiffnesses are a bit less important than K55, the resulting stiffness can still be used in some work:
[image]

ptrbortolotti, 2023-03-08:
good job. a log y plot or a diff plot will better highlight the differences between the two approaches. @fzahle at DTU noticed differences up to 10% or so. I'm sure natural frequencies are also impacted. in this branch https://github.com/ptrbortolotti/SONATA/tree/adhesive I switched from a full TE made of uni glass to a full TE made of adhesive. That should help reducing the gap between SONATA and BECAS

closed 2023-03-20 by MassimoSirigu