IEAWindSystems/IEA-15-240-RWT #126: Theory behind blade optimisation in Wisdem

MassimoSirigu, 2023-03-20:
Good morning,
I would like to ask you some questions about the structural optimisation of the blade performed in Wisdem: during the optimisation, is it true that the only stress considered is the longitudinal stress? transverse and shear stress are neglected? Is it performed using 2D sections or a complete shell model, like Numad, that uses Ansys APDL? Are the buckling effects considered? for the final evaluation of the AEP, is it considered the twist deformation (for example, using Beamdyn stand-alone), that reduces the power output? 
I noted that the use of Beamdyn reduces significantly the power output, I think that the reduction (or elimination) of the thrust peak shaving coefficient can also have a positive effect to enhance power production.
In the image, the power curve using Beamdyn for the floating wind turbine, using a wind shear exponent of 0.14, turbulence model class B, and baseline control parameters for 20 minutes of simulation. the mean wind speed is taken at 100 meters above sea water level.
[image]

ptrbortolotti, 2023-03-21:
Hello,

the initial design did not include a detailed FEM analysis, did not include buckling constraints, and did not include torsional deformations. Strains were computed along the spar caps using this formula https://github.com/WISDEM/WISDEM/blob/c07ce084b582322fa2e73da70ce2396abecf6e7d/wisdem/rotorse/rotor_structure.py#L702, which includes axial force and the two moments along the principal axes.

If you are looking for detailed structural redesigns of the blades, I'd recommend checking out two very relevant studies from the University of Bristol and The University of Texas at Dallas. The respective papers are listed in the README of the develop branch
https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/develop

Lastly, torsional deformations will certainly impact the power curve. However I believe not as badly as your predictions. Were you prescribing the pitch angle? Or were you letting ROSCO adjust to maximize power? Also, as you say, peak thrust shaving will certainly impact power.

I hope this helps.
Regards,
Pietro

MassimoSirigu, 2023-03-21:
Hello, I downloaded recently the IEA 15 MW and OpenFAST 3.4.0 and I did not modify the parameters (structure or control). The generator DOF is true and the ROSCO controller defines the blade pitch and torque. If you say that my results are not as you would expect, I have probably made some mistakes, I will check the wind scatter or beamdyn implementation again.

MassimoSirigu, 2023-03-21:
Can I ask you what is the difference between the trapezoidal and gaussian quadrature: from the beamdyn manual, "Trapezoidal quadrature is appropriate when the number of “blade input stations” (described below) is significantly greater than the order of the LSFE". 21 sections, as defined in baseline beamdyn implementation, are sufficient for 10th order of interpolation? Or should use gaussian interpolation? what are the drawbacks of the gaussian quadrature?

michaelasprague, 2023-03-21:
@MassimoSirigu  The advantage of Trapezoidal Quadrature is that it creates the BeamDyn stiffness and mass matrices by sampling (at least) at the points where the 6x6 sectional matrices are provided.   When using Gaussian Quadrature, one will often "miss" much of the information in the sectional matrices.  However, for Trapezoidal Quadrature, one must use enough integration points to (a) capture the material properties, and (b) integrate the basis functions with sufficient accuracy.  21 sections is likely sufficient, but one can do a check by doubling that number.

ptrbortolotti, 2023-03-21:
The behavior is probably correct then, is the pitch angle varying in those 50 seconds? We did not account for torsional deformations during the conceptual design phase and you are correct that they do impact power significantly

MassimoSirigu, 2023-03-21:
here you have the blade pitch variation over time. the blade pitch is saturated due to the thrust peak shaving.
Thank you for your support.
[image]

closed 2023-03-21 by MassimoSirigu