IEAWindSystems/IEA-15-240-RWT #127: Invalid logical input for file "IEA-15-240-RWT-UMaineSemi.fst" occurred while trying to read SumPrint.

maucollu, 2023-03-23:
## Description
Hi,
I downloaded the IEA-15-240-RWT model of the UMaine semisub.

I also downloaded the pre-compiled version of OpenFAST 3.0.0 from here: (as far as I read, this is the most up-to-date version that can be used with this model?), as well as the file MAP_x64.dll

When trying to launch the simulation with the default .fst, I got the following error:

FAST_InitializeAll:FAST_Init:FAST_ReadPrimaryFile:Invalid logical input for file
"IEA-15-240-RWT-UMaineSemi.fst" occurred while trying to read SumPrint.

### Steps to reproduce issue

1. Downloaded  openfast_x64.exe and MAP_x64.dll  pre-compiled from here: [https://github.com/OpenFAST/openfast/releases/tag/v3.0.0](url) 
2. Downloaded the IEA-15-240-RWT folder from here: [https://github.com/IEAWindTask37/IEA-15-240-RWT](url)
3. Unzipped the IEA-15-240-RWT, copied openfast_x64.exe and MAP_x64.dll in IEA-15-240-RWT-master\IEA-15-240-RWT-master\OpenFAST\IEA-15-240-RWT-UMaineSemi 
4. Run `openfast_x64.exe IEA-15-240-RWT-UMaineSemi.fst`

### Current behavior
**************************************************************************************************
OpenFAST

Copyright (C) 2021 National Renewable Energy Laboratory
Copyright (C) 2021 Envision Energy USA LTD

This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
See the "LICENSE" file distributed with this software for details.
**************************************************************************************************

OpenFAST-v3.0.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 1900
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jun 23 2021
 - Time: 21:25:05
Execution Info:
 - Date: 03/23/2023
 - Time: 16:28:22-0000

OpenFAST input file heading:
    IEA 15 MW offshore reference model on UMaine VolturnUS-S semi-submersible floating platform

FAST_InitializeAll:FAST_Init:FAST_ReadPrimaryFile:Invalid logical input for file
"IEA-15-240-RWT-UMaineSemi.fst" occurred while trying to read SumPrint.

 OpenFAST encountered an error during module initialization.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.

### Expected behavior
A successfully completed simulation

### Code versions
OpenFAST-v3.0.0

gbarter, 2023-03-24:
Per the release notes, the input files in the `master` branch are consistent with OpenFAST v3.3.  We are prepping input files to be consistent with the latest OpenFAST release in the `develop` branch.

maucollu, 2023-03-24:
Thanks a lot - sorry, I was suspecting I missed something basics, since no one was complaining about this....
Only one thing: in the main page of the github, it mentions, under "Requirements":
"Requirements

OpenFAST:

    The OpenFAST input files are compatable with OpenFAST-v3.0. This can be compiled [from source here](https://github.com/OpenFAST/openfast.git) or precompiled Windows binaries are [available for download](https://github.com/OpenFAST/openfast/releases/latest/download/windows_openfast_binaries.zip). More information on installing and running OpenFAST is available in [OpenFAST documention](https://openfast.readthedocs.io/en/master/)."

Kind Regards

Maurizio