IEAWindSystems/IEA-15-240-RWT #132: External Stiffness matrix - WAMIT 2nd Order

mgjat, 2023-03-30:
Hi, 

I just want to make sure that the "Global external stiffness matrix" used in the WAMIT-2nd Order input file (.frc) from K03-15MW-semi-sub_UMaine_OpenFAST corresponds to the mooring system and if it takes into account the platform orientation. 

It is set as:

    1	ISTIF
    6.36E+04	0.00E+00 0.00E+00	0.00E+00	-1.47E+05	0.00E+00
    0.00E+00	6.36E+04	0.00E+00	1.47E+05	0.00E+00	0.00E+00
    0.00E+00	0.00E+00	2.10E+05	0.00E+00	0.00E+00	0.00E+00
    0.00E+00	1.47E+05	0.00E+00	1.69E+08	0.00E+00	0.00E+00
    -1.47E+05	0.00E+00	0.00E+00	0.00E+00	1.69E+08	0.00E+00
    0.00E+00	0.00E+00	0.00E+00	0.00E+00	0.00E+00	5.37E+07

Does this external stiffness matrix corresponds to the IEA-15-240-RWT-UMaineSemi mooring system?

 Thanks for your help.

gbarter, 2023-04-05:
Thank you for the question.  @luwang00 is looking into this, but we will need a bit more time to come back to you with a formal response

luwang00, 2023-04-13:
Hi @mgjat, based on what @mattEhall dug up, we believe the external stiffness matrix in the WAMIT 2nd-order input file is indeed intended to represent the linearized mooring stiffness. However, we are currently having some trouble reproducing the numbers with the mooring setup defined in the corresponding MoorDyn input file. We will try to recompute the mooring stiffness matrix and do some comparison with it. In the meantime, please be aware of Issue #134 in case you plan to use the QTF files.

luwang00, 2023-04-23:
Hi @mgjat,

Sorry for the delay, but in case you are still interested in this, we have recomputed the linearized mooring stiffness matrix using the latest definition of the mooring setup. 

**Platform at the reference position:**
```
+7.192e+04      +0.000e+00      -3.587e-01      +0.000e+00      +1.145e+06      +0.000e+00
+0.000e+00      +7.192e+04      +0.000e+00      -1.145e+06      +0.000e+00      +2.165e+01
-3.587e-01      +0.000e+00      +6.076e+04      +0.000e+00      -3.198e+00      +0.000e+00
+0.000e+00      -1.145e+06      +0.000e+00      +2.587e+08      +0.000e+00      +3.995e+02
+1.145e+06      +0.000e+00      -3.198e+00      +0.000e+00      +2.587e+08      +0.000e+00
+0.000e+00      +2.165e+01      +0.000e+00      -1.908e+01      +0.000e+00      +2.524e+08
```
**Platform with 10 m offset in surge (due to, e.g., wind thrust):**
```
+9.251e+04      +0.000e+00      +1.271e+04      +0.000e+00      +1.273e+06      +0.000e+00
+0.000e+00      +6.063e+04      +0.000e+00      -1.030e+06      +0.000e+00      -1.502e+04
+1.271e+04      +0.000e+00      +6.135e+04      +0.000e+00      +7.885e+04      +0.000e+00
+0.000e+00      -1.030e+06      +0.000e+00      +2.411e+08      +0.000e+00      -2.500e+07
+1.273e+06      +0.000e+00      +7.885e+04      +0.000e+00      +2.873e+08      +0.000e+00
+0.000e+00      -1.502e+04      +0.000e+00      -1.293e+07      +0.000e+00      +2.626e+08
```
The units are N, m, and radian. All moment and rotation are about the platform reference point at the center of the waterplane area.

Also, we have just created a pull request #137 to address issue #134. Note that the updated second-order wave-excitation QTFs are still based on the original stiffness matrix you quoted in your original post. We might rerun the second-order solver with the updated stiffness matrix above later on.

mgjat, 2023-04-24:
Thanks so much for the information @luwang00. 
I will work with the first order.