IEAWindSystems/IEA-15-240-RWT #135: CoG of platform is changed from the report

MassimoSirigu, 2023-04-06:
Good morning,
I noted that some of the variables of the floating platform are different between the file in Github and the report https://www.nrel.gov/docs/fy20osti/76773.pdf.
For example, the center of mass is -14.4m and in the report is 14.94m, while other parameters ( displaced volume, platform inertia and mass) are the same (github mass =17838 ton and report mass = 17854).
I ask you why the cog is changed and also if the geometry parameters that define the hydrodynamic response are changed.
Thank you,
Massimo Sirigu

gbarter, 2023-04-06:
Hello Massimo,

You can find this issue addressed in the FAQ:
https://github.com/IEAWindTask37/IEA-15-240-RWT/wiki/Frequently-Asked-Questions-(FAQ)#why-do-the-simulated-volturnus-s-properties-differ-from-the-report

Best,
Garrett

Junruzhang627, 2024-03-15:
Hi @gbarter I am trying to understand the pyhsical meaning of the floater COG. I have hand-calculated the floater COG by only include the mass of mooring. But I get the wrong value which is not the value defined in the report. It seems that the floater COG not only include the influence of the mooring, but include the influence of other object. I don’t know what the other object is. Do you have any idea about this? Can you explain for me the algorithm of calculating the floater COG?

gbarter, 2024-03-15:
Perhaps this [document](https://www.nrel.gov/docs/fy14osti/60601.pdf) would help?  Platform CG typically does not include the mooring lines.  Furthermore, the variable ballast is set such that all of the forces balance.  When accounting for the mooring lines, you have to compute the force on the platform, which is not the mass of the mooring lines since their weight in water is different and some of the mooring line length lays on the seabed.

Junruzhang627, 2024-03-15:
> Perhaps this [document](https://www.nrel.gov/docs/fy14osti/60601.pdf) would help? Platform CG typically does not include the mooring lines. Furthermore, the variable ballast is set such that all of [...]

Hi @gbarter, thank you very much for providing me material and advice. Can you send me the similar material of IEA 15MW floating material with the detailed modeling information of the floater?

Junruzhang627, 2024-03-15:
> Perhaps this [document](https://www.nrel.gov/docs/fy14osti/60601.pdf) would help? Platform CG typically does not include the mooring lines. Furthermore, the variable ballast is set such that all of [...]

Hi @gbarter, in the material, it mentioned that "The CM of the floating platform, which includes **everything except the tower,
rotor-nacelle assembly, and moorings,** is located at 13.46 m along the platform centerline below the SWL". In this paper, whether "everything" means only the floater plotted in the below figure is considered?
[image]

gbarter, 2024-03-15:
Yes, the floater as pictured is what is considered for the CM calculation.  For the ballasts, I am unsure if the reported CM from that paper includes the variable water ballast, but it likely is since ballast was included in the reported total mass value.  If it helps to see the OpenFAST files for the DeepCwind semisub, they are [here](https://github.com/OpenFAST/r-test/blob/main/glue-codes/openfast/5MW_OC4Semi_WSt_WavesWN/NRELOffshrBsline5MW_OC4DeepCwindSemi_HydroDyn.dat).

For the 15MW semisub, there is no other document describing the geometry besides the report.

Junruzhang627, 2024-03-18:
@gbarter thank you very much for your help. I will try to hand-calculate the floater cm according to the HydroDyn files you provided for me.

closed 2024-07-11 by gbarter