IEAWindSystems/IEA-15-240-RWT #136: A request for OpenFOAM case files and WAMIT input files

hachikoi1, 2023-04-19:
# Description of feature
Dear Developers
Could you please provide me the OpenFOAM case files and WAMIT input files of UMaine VolturnUS-S platform?

Very thankful
Best regards,

# Potential solution
Can you think of ways to implement this?

gbarter, 2023-04-19:
We do not currently support OpenFOAM analysis of the reference turbine.  If you would like WAMIT input files, please see: https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/HydroData

hachikoi1, 2023-04-19:
> We do not currently support OpenFOAM analysis of the reference turbine. If you would like WAMIT input files, please see: https://github.com/IEAWindTask37/IEA-15-240-RWT/tree/master/OpenFAST/IEA-15-240-RWT-UMaineSemi/HydroData

Dear [gbarter](https://github.com/gbarter)
I found some analysis of UMaineSemi platform in https://www.nrel.gov/docs/fy20osti/76773.pdf using OpenFOAM. Do you have any OpenFOAM cases of the platform?
Best regards,

gbarter, 2023-04-19:
Unfortunately that was a one-off activity and those analysis files are no longer available

hachikoi1, 2023-04-22:
> Unfortunately that was a one-off activity and those analysis files are no longer available

Dear [gbarter](https://github.com/gbarter)
I want to calculate the quadratic drag at platform COG. Could you please provide the drag coefficient Cd for each platform structure and provide the formula for calculating quadratic drag (especially in the Roll, Pitch and Yaw direction)?

gbarter, 2023-04-24:
That info is also not readily available, sorry. You would have to do some analysis to compute the values you are looking for.

hachikoi1, 2023-04-24:
That is ok. Thank you very much for your reply

gbarter, 2023-04-24:
Please take a look at this question response:
https://forums.nrel.gov/t/iea-15mw-volturnus-s-mean-platform-pitch/4829/2

closed 2023-05-30 by hachikoi1