IEAWindSystems/IEA-15-240-RWT #139: Quadratic drag matrix AddBQuad in HydroDyn

hachikoi1, 2023-05-10:
# Description of feature
Describe the feature here and provide some context. Under what scenario would this be useful?
Dear all,
I am very confused about the quadratic drag matrix AddBQuad in HydroDyn. I expected that the non-diagonal elements should all be zero, but (1,5), (5,1), (2,4), (4,2) in AddBQuad are non-zero. Could anyone give me some guidance on how to calculate AddBQuad, especially for the area A in roll, pitch and yaw directions. 
Moreover, the reference point for AddBQuad is at platform reference point. How to calculate the AddBQuad at COG?

# Potential solution
Can you think of ways to implement this?

gbarter, 2023-05-11:
I believe you might have received a direct e-mail reply about this question, but let me know if that isn't the case.

hachikoi1, 2023-05-12:
Dear [gbarter](https://github.com/gbarter)
I used to be very confused about the calculation formula for AddBQuad. After talking about AddBQuad with [Jason](https://forums.nrel.gov/t/hydrodynamic-drag-loads-formulation/2317/31) and developers of WEC-Sim (calculates quadratic drag based on the reference point at platform COG), I found the AddBQuad in HydroDyn is the fitted result based on the calculation difference between OpenFAST (without viscous effect) and OpenFOAM (with viscous effect). Am I right? 
Theoretically, we can get quadratic drag and velocity at COG based on the the quadratic drag, displacement and velocity at PRP. By solving the equation system of F_{quaddragCOG} = -0.5CdrhoAv_Cog|v_Cog| at each time step, we can ontain quadratic drag coefficient at COG. Unfortunately, the resulting AddBQuad at COG obtained at each time step is different.
I'm trying to compare the results with and without the viscosity effect to get AddBQuad at COG. Thus, I wonder to know the method of viscosity correction.  **Is the viscosity correction done separately for each degree of freedom, or the correction is based on the decay test results for only one degree of freedom, e.g. Ry?**

luwang00, 2023-05-12:
Hi @hachikoi1,

I'll try to answer some of your questions below, but the short answer is that there's no single "correct" approach to obtaining the quadratic damping coefficients.

The AddBLin and AddBQuad coefficients in HydroDyn are all intended as tools the user can use to add more motion damping into the model beyond what's captured by potential-flow wave-radiation damping. Yes, they can be used to represent the viscous damping not captured by potential-flow theory, but they can be used for other purposes as well.

At a high level, viscous loading is highly complex and sensitive to the flow conditions. The quadratic damping model is only a highly simplified representation of this complex loading, and one can only hope to approximate the overall behavior of the system with this simplified model for a given condition. It is not uncommon for uniform current, free-decay motion, and wave conditions (which also depend on the wave height and frequency) to give very different estimates of the quadratic damping coefficients. Therefore, there is no "correct" method or formula you can use to "calculate" these coefficients, and there's no simple answer to your question.

In your original post, I'm guessing you are referring to the quadratic damping matrix in the VolturnUS reference document. The non-zero coupling terms come from the fact that the motion and load are all referenced about the platform reference point on the still water level. The easiest way to visualize this to imagine the structure translating steadily in the surge direction. The line of action of the resulting drag force on the submerged section of the platform is somewhere below the still water level, which will create a pitch moment about the platform reference point above. This is the (4,1) coupling term. However, keep in mind that the coefficients in the reference document were only rough estimates from OpenFOAM CFD simulations with constant fluid velocity. Therefore, there's no guarantee these coefficients can be used for any other conditions, so you should not treat these as the "correct" values.

At the end of the day, the user need to choose the coefficients in order to obtain reasonable motion response for the conditions in question, and there's no single correct approach to achieve this.

closed 2023-05-30 by hachikoi1